thonny/thonny #2935: tooltips - can I get rid of them

mikedabb6, 2023-09-20:
I have the same issue as others with the tool tips not going away.

I have Thonny version 4.0.1. installed.

Can you please tell me how to install Version 4.1.2 on my Raspberry Pi 400.

Thanks in advance

aivarannamaa, 2023-09-21:
You can either `apt remove thonny` and then `pip install thonny` or you can wait until RPi people update their repositories.

I have plans to release 4.1.3 during this weekend and after this I could poke RPi package managers to make the update.

mikedabb6, 2023-09-21:
Hi Thanks for the quick reply.I will look into apt remove and reinstall and see if that works.Thanks for poking the RPi people if that doesn’t work.RegardsMikeSent from my iPhone

closed 2023-11-19 by aivarannamaa