thonny/thonny #2940: Open System Shell command closes pi peco connection

rvwilliams88, 2023-09-25:
I'm running Thonny 4.1.2 on Windows 11.  I want to install a package using pip.  When I click on the 'Open system shell' command it closes the connection to my pi peco w and gives me a Windows prompt. The two screenshots show before the command and after it.
[image]
[image]

aivarannamaa, 2023-09-27:
Thank you for the report!

The command name "Open system shell" is probably misleading in MicroPython mode. In this context the word "system" represents the MicroPython runtime and this command closes the serial connection in Thonny and reopens it in a serial command line program. The connection must be closed in Thonny, because only one process can have serial connection to a device at a time.

aivarannamaa, 2023-09-27:
If you want to install a MicroPython package on your device, then you could use "Tools => Manage packages", which re-uses Thonny's serial connection to the device.

rvwilliams88, 2023-09-27:
Hi Aivar

Thanks for the response.  I couldn't use the package manager because of the expired certificate issue.  I was following the instructions in the Thonny help :

1. From "Tools" menu select "Open system shell...". You should get a new terminal window stating the correct name of pip command (usually pip or pip3). In the following I've assumed the command name is pip.
2. Enter pip install <package name> (eg. pip install pygame) and press ENTER. You should see pip downloading and installing the package and printing a success message.
3. Close the terminal (optional).
4. Return to Thonny.

How is the above meant to work on a Windows-based PC?

Regards

Richard Williams.

****************************************************
Richard Williams
email: ***@***.******@***.***>
mobile: 07981 734697
****************************************************

aivarannamaa, 2023-09-27:
These instructions are for installing packages for CPython, not for MicroPython.

As a work-around, you could download the required package manually with a web-browser, uncompress and transfer the files to your MicroPython device's `/lib` folder using Thonny's file browser (View => Files).

rvwilliams88, 2023-09-27:
Hi Aivar, thanks for the explanation. Pity that isn't clearer in the Help.
Regards
Richard Williams

closed 2023-11-13 by aivarannamaa