thonny/thonny #2941: Thonny py5 crashes after mac software update

treinartz, 2023-09-27:
the new macintosh update prevents the py5 popup from remaining open in Thonny, it seems to crash or be open in another window..it may have to do with the stage manager, but I am not sure

aivarannamaa, 2023-09-28:
Thank you for the report!

Unfortunately I don't know what a py5 popup is. Can you give me the steps for reproducing the problem?

treinartz, 2023-09-28:
Sure..when you run a pt5 sketch typically a window pops up with the
drawing.  After the Mac update, in some cases, the window pops up the
immediately goes away.  Some have fixed the issue by turning off the stage
manager, but for others the problem persists.

Thanks for your help!

aivarannamaa, 2023-09-28:
Does py5 work properly with new macOS when you run it without Thonny?

treinartz, 2023-09-28:
I have not checked that…it may just be the stage manager trying to manage
the canvas window since in some cases it works when the stage manager is
off….I’ll gather more information in the meantime

treinartz, 2023-09-28:
On the Mac, it looks like the following code needs to be added to get
the py5 window to appear: After setup()

def setup():
    size(400, 400)

add this code before calling other functions

def draw():
    pass

at least that seems to be a start....

aivarannamaa, 2023-09-28:
I believe this issue is best taken to the py5 team: https://github.com/py5coding/py5generator/discussions

closed 2023-09-28 by aivarannamaa

treinartz, 2023-09-28:
got it, thank you!