thonny/thonny #2943: thonny 4.1.3 source code actually is for version: 5.0.0b1.dev0

algent-al, 2023-09-29:
I maintain `thonny` for Solus.
While updating `thonny` using source code `https://github.com/thonny/thonny/archive/refs/tags/v4.1.3.tar.gz`, I noticed that the version is 5.0.0b1.dev0 and not 4.1.3.

aivarannamaa, 2023-09-29:
Oops! Thank you for the notification! 

I forgot to mind the branch name when tagging the release. I now deleted tag `v4.1.3` and re-created in `v4_1` branch.

BTW, have you considered using tarballs at PyPI (e.g. https://files.pythonhosted.org/packages/86/2b/7ffd80b5413859bddc93eb6c1256091dfc529cac8ac3b8a1764e57618d63/thonny-4.1.3.tar.gz) as the base for your package? I would consider these more "official" than the tarballs at GitHub. 

For easier Debian packaging I also maintain a subset of these tarballs at https://thonny.org/tarballs/

algent-al, 2023-09-29:
Yes, I already updated `thonny` on Solus by using PyPI tarball.

closed 2023-09-29 by aivarannamaa