thonny/thonny #2944: Save shell window to a text file

EricMyers47, 2023-09-30:
I would like to be able to easily save the output that shows up in the Shell window to a text file with a single pull-down menu item. 

When my students submit an assignment they give me both the script (.py file) and the output (.txt file, and sometimes .jpg/.png if there is graphics).    With IDLE they can select the separate shell window and then File -> Save as...

This would probably require a new File menu item, something like "Save shell"  or "Save shell as..."  (and maybe then "Save as..." becomes "Save script as..."?)

I do have a workaround:  Select all text in the shell window (ctrl-A or ⌘-A), then "File -> New, then paste into the new file (ctrl-V or ⌘-V) and Save or Save as..., taking care to change the file extension to .txt  (since it defaults to .py).

For the shell window the default extension could be .txt or .log  (I'm not sure which is the better choice).

This is similar to #682 but would save to a .txt file instead of printing.   In fact, it solves #682 because once one can save the shell output to a file then that can be printed, and then you don't have to do the work to interface directly with different printing systems on different OS's.

aivarannamaa, 2023-09-30:
Thank you for the suggestion!

I'd rather not rename "Save" and "save as", but I could add a command to Shell's right-click menu.

EricMyers47, 2023-10-01:
Whatever you think is best.   Adding it to that window makes it clear that it applies to that window only.

I'm not suggesting changes to "Save" or "Save as..."     Those should work as they have before, on the code only, as any user would expect. But one more File menu item at the bottom, clearly away from the others, could be "Save Shell as...".    

There is a precedent for something like this in another tool with multiple windows that my students used, called Logger Pro (from Vernier.com).  It's used for collecting and graphing data in a lab course.   The "Print" menu prints both the graph and the data table, but they've also added "Print Graph..." and "Print Data Table..."    (see this screen shot). 

[image]

(Since Logger Pro uses Print not Save they have the problem of dealing with different print systems, and in fact printing to PDF on Windows is broken and has been for a couple of years.  Sigh.)

But anything is better than nothing, so if you start with the right click menu for the shell window it would be a useful improvement.   

I should mention that I've been directing my students to IDLE over the past few years even though I knew about Thonny from Rpi.  Now that it's easy to install on Windows and Mac I'm adding Thonny as an option.  Thank you for this tool, and for thinking about this suggestion.