thonny/thonny #2947: Can't install package

drsm79, 2023-10-02:
Hi,
I'm trying to install a package to a raspberry pi pico. I get the following error (seems independent of the package):

```
install --progress-bar off micropython-ssd1306
PROBLEM IN THONNY'S BACK-END: Exception while handling 'install_distributions' (AssertionError).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 1.
```

Followed by:

```
Traceback (most recent call last):
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/tkinter/__init__.py", line 1921, in __call__
    return self.func(*args)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/pip_gui.py", line 355, in _on_install_click
    self._perform_pip_action("install")
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/pip_gui.py", line 818, in _perform_pip_action
    if self._has_remote_target():
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/pip_gui.py", line 1059, in _has_remote_target
    return proxy.supports_remote_files()
AttributeError: 'NoneType' object has no attribute 'supports_remote_files'
```

Backend log has:

```
[MainThread] ERROR   thonny.backend: Exception while handling 'install_distributions'
Traceback (most recent call last):
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/backend.py", line 285, in _handle_normal_command
    response = handler(cmd)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/micropython/mp_back.py", line 960, in _cmd_install_distributions
    assert not any([arg.startswith("-") for arg in args])
AssertionError
```

And then the board disconnects. Any suggestions? I can write files to the board, the REPL works etc.

drsm79, 2023-10-02:
Oh, this is the Thony version info: [image]

drsm79, 2023-10-03:
I can install the module via the pipkin CLI.

humphrey13, 2023-10-04:
i think I have the same problem

springjools, 2023-10-05:
I managed to work around it by finding a rpi pico with an already installed package, and copying it over with thonny's view files mode.

gumbald, 2023-10-06:
> I can install the module via the pipkin CLI.

I have the same problem but not sure what pipkin is - can you give me some tips here please? Does a previous version of Thonny work or is this server-side?

tenapier, 2023-10-06:
Yes, I had the same problem. Could not successfully download (micropython-bme280) from package manager in Thonny. But, using the terminal, and downloading pipkin, (https://pypi.org/project/pipkin/) , and then using it in the command line to download micropyton-bme280 to my RP Pico W worked! Good luck!

springjools, 2023-10-07:
Do you install pipkin with pip (or mip) to thonny or to any other python installation? I mean does it communicate with pico by itself, or does it need thonny? Sorry if asking a question, i tried looking at the docs.

drsm79, 2023-10-07:
I'm now using vscode (following [this guide](https://www.hackster.io/shilleh/how-to-use-vscode-with-raspberry-pi-pico-w-and-micropython-de88d6)) & got libraries installed with pipkin as follows:

- made a virtualenv `python3.10 -m venv venv`
- activated it `source venv/bin/activate`
- installed pipkin (on laptop, in the venv) `pip install pipkin`
- installed the package I wanted to the board `pipkin install micropython-ssd1306`

HTH!

tenapier, 2023-10-07:
> Do you install pipkin with pip (or mip) to thonny or to any other python installation? I mean does it communicate with pico by itself, or does it need thonny? Sorry if asking a question, i tried looking at the docs.

 pip install pipkin
 pipkin --port /dev/cu.usbmodem14601 install micropython-bme280

While installing (second line) I quit Thonny (so that port would not appear in use) and identified the port that the RPPicoW was on (because I have other ports open on my Mac), and it seem to work just fine. Hope this helps and adds to the discussion.

pleumann, 2023-10-08:
Same here on MacOS Monterey. Older versions 4.0.1 and 3.3.13 work fine.

Sonic557, 2023-10-19:
I got the same issue here on all my students' PCs and Macs. Older version 4.0.2 works fine. Bumping up for dev's attention.

lark-cig, 2023-10-25:
I have the same problem ( Mac  12.6.2) Can't import any packages. 

```
13:57:23.311 [MainThread] INFO    thonny: TIME/MODS 219.884 before install_distributions 
13:57:23.317 [MainThread] ERROR   thonny.backend: Exception while handling 'install_distributions'
Traceback (most recent call last):
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/backend.py", line 285, in _handle_normal_command
    response = handler(cmd)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/micropython/mp_back.py", line 960, in _cmd_install_distributions
    assert not any([arg.startswith("-") for arg in args])
AssertionError
```

When I click on the link "here" to install from local file, Thonny just quits. When restarting Thonny it says "Unable to connect" and I have to disconnect the USB. The log shows nothing.

wanderingstan, 2023-10-27:
Bumping @Sonic557 's comment. Downgrading to [version 4.0.2](https://github.com/thonny/thonny/releases/tag/v4.0.2) fixes the issue if you just want to get moving.

springjools, 2023-10-27:
Don't people test their software before releasing it? But thanks for the workaround 👍

matteius, 2023-11-06:
This is affecting my project as well:  https://github.com/opensensor/growmax/discussions/5

aivarannamaa, 2023-11-18:
Thank you for the report and sorry for the late reply!

The problem will be fixed under #2979

closed 2023-11-18 by aivarannamaa

aivarannamaa, 2023-11-19:
Please try https://github.com/thonny/thonny/releases/tag/v4.1.4

Redika-ak, 2024-03-29:
> Please try https://github.com/thonny/thonny/releases/tag/v4.1.4

still, I can't update the plugin

SalisStreeg, 2024-06-06:
Hello, i had the same problem.
And i just switcht from python interpreter to Thonny's an that workt for me.
(sorry for my English)

tenapier, 2024-06-06:
How were you able to switch interpreters from your Python's to Thonny's
python interpreter?
Thanks,
Tim

SalisStreeg, 2024-06-24:
Sorry for the late response, I didn't see I had mail. I can't check right
now cause I'm on my phone but I thought to just let you know I saw your
email.

Suresh6060, 2024-09-28:
I have a similar problem. I on windows 7 , thonny version  4.1.4 using rpi pioc w  pyhton 3.8.10 32 bit

I wanted to install html library

in the "manage packeges" html shows as from html @micropython-lib
further clicking on it , shows 
Latest stable version  3.3.4
License : MIT 

however when install button is clicked it shows error
 install --progress-bar off html
Error 'type' object is not subscriptable

The interpreter is MicroPython(Raspberry pi pico) 

I tried changing to MicroPython (RP2040) and MicroPython(generic) . The same error appears. 

Earlier I had tried installing picozero using inpreter LocalPyton 3. It installed could not use it in the code due to library not found error. 

Is there any solution. I am not very experienced in python. Is there any simple method for this issue. Or is it unsolvable ?