thonny/thonny #2949: PICO lightsleep causes Thonny to hang

davefes, 2023-10-04:
As per discussion:
https://github.com/orgs/micropython/discussions/12573
Thonny appears to hang after 5-20 loops of the following program:

from time import sleep
from machine import lightsleep

while True:
    print ('waiting 2 seconds for a CTRL-C')
    sleep(2)
    print ('going to lightsleep for 2 seconds')
    sleep(0.1)
    lightsleep(2000)
    print ('woke-up again')
    sleep(0.1)

lightsleep messing with the USB has been suggested as a possible cause.

aivarannamaa, 2023-11-16:
Thank you for the report and sorry for the late reply!

Have you found out more about this problem meanwhile?

davefes, 2023-11-16:
To be honest I found trying to get the Pico to do the things that I had accomplished on the ESP32 platform difficult.  So, I migrated to the STM32F411 BlackPill for my non-WiFi applications.