thonny/thonny #2950: Packages not working.

DV3Dprintservice, 2023-10-04:
### i think its a bug but i dont know??
I just instalt Thonny _4.1.3_ on my Win 11 pc and whent and downlode the ''ssd1306'' package but i got a error.(see first photo)

when i got to some ather package i got a board error.(see second photo)

[image]
[image]

VEMA-ANESH, 2023-10-12:
Iam also facing same issue

aivarannamaa, 2023-11-18:
Thank you for the report and sorry for the late reply!

The second error will be handled under #2979. I can't get 404 with the package you mentioned.

aivarannamaa, 2023-11-19:
Please try https://github.com/thonny/thonny/releases/tag/v4.1.4