thonny/thonny #2959: Unable to connect with RPPicoW once main.py is run on it

tenapier, 2023-10-13:
Unable to connect to /dev/cu.usbmodem14601: [Errno 2] could not open port /dev/cu.usbmodem14601: [Errno 2] No such file or directory: '/dev/cu.usbmodem14601'

This is what I get in Thonny, when I try to use my RPPicoW after it has a main.py program installed and having run. The only workaround is with flash_nuke.uf2 and re-install RPI_PICO_W-20231005-v1.21.0.uf2, and start all over again. Hoping you can get this issue fixed. Am using Thonny 4.1.3 on an Intel Mac with macOS Monterey v 12.7 . Thanks for any help!

aivarannamaa, 2023-10-14:
Thank you for the report!

What do you have in your main.py? Does the device appear at another port after you unplug and re-plug it?

tenapier, 2023-10-14:
It's a program to monitor temp, barometric pressure and humidity from a bme280, with date and time, and I am trying now to put a deepsleep function in as well. Would like to have a standalone solar connected unit to monitor some of the weather.

No, I haven't noticed that the device comes up at another port site (but didn't specifically check that).

Interestingly, I just plugged in a PicoW with a main.py on it, and it did come up okay in my Thonny program. That wasn't happening before, and for quite a while. Not sure if it is fixed or stable, or anything changed. Can let you know after I use it for a while today.

Thank you for your reply, and help!
Tim

#import modules
import network
import socket
import time
import utime as time
from utime import sleep
from machine import Pin, I2C, WDT
from machine import RTC
from machine import deepsleep
import bme280
import rp2
import sys

import usocket as socket
import ustruct as struct

ssid = '**************' #Your network name
password = '**************' #Your WiFi password

# wintertime / Summerzeit
#GMT_OFFSET = 3600 * -6 # 3600 = 1 h (wintertime)
GMT_OFFSET = 3600 * -5 # 3600 = 1 h (summertime)

# NTP-Host
NTP_HOST = 'pool.ntp.org'

#initialize I2C 
i2c=I2C(0,sda=Pin(0), scl=Pin(1), freq=400000)

def getTimeNTP():
    NTP_DELTA = 2208988800
    NTP_QUERY = bytearray(48)
    NTP_QUERY[0] = 0x1B
    addr = socket.getaddrinfo(NTP_HOST, 123)[0][-1]
    s = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
    try:
        s.settimeout(5)
        res = s.sendto(NTP_QUERY, addr)
        msg = s.recv(48)
    finally:
        s.close()
    ntp_time = struct.unpack("!I", msg[40:44])[0]
    return time.gmtime(ntp_time - NTP_DELTA + GMT_OFFSET)

def setTimeRTC():
    rtc = RTC()
    tm = getTimeNTP()
    rtc.datetime((tm[0], tm[1], tm[2], tm[6] + 1, tm[3], tm[4], tm[5], 0))
    
def connect():
    #Connect to WLAN
    wlan = network.WLAN(network.STA_IF)
    wlan.active(True)
    wlan.config(pm = 0xa11140)  # Disable powersave mode
    wlan.connect(ssid, password)
    while wlan.isconnected() == False:
        print('Waiting for connection...')
        sleep(1)
    ip = wlan.ifconfig()[0]
    print(f'Connected on {ip}')
    return ip

def open_socket(ip):
    # Open a socket
    address = (ip, 80)
    connection = socket.socket()
    connection.bind(address)
    connection.listen(1)
    return connection

def webpage(temp,pressure,humidity,now0,now1,now2,now4,now5):
    #Template HTML
    html = f"""
            <!DOCTYPE html>
            <html>
            <head>
            <title>Pico W Weather Station</title>
            <meta http-equiv="refresh" content="5">
            </head>
            <h1 <br><br> </h>
            <h2 style="background-color:DodgerBlue; color:White; font-size:350%;"> <center>Weather App</center></h>
            <body>
            <p style= "font-size:200%;">   <br><br>    </p>
            <p style="font-size:100%; color:White; background-color:Tomato;">Temperature: {temp} &#176;F<br><br>Barometric Pressure: {pressure} inHg <br><br> Humidity: {humidity}</p>
            <p style="font-size:100%; color:White; background-color:DodgerBlue;">Date: {now1}-{now2}-{now0}</p>
            <p style="font-size:100%; color:White; background-color:DodgerBlue;">Time: {now4}:{now5}</p>
            </body>
            </html>
            """
    return str(html)
    
def serve(connection):
    #Start a web server
     while True:
        
        for i in range(5):
           bme = bme280.BME280(i2c=i2c)
           temp = bme.values[0]
           temp = float(temp)
           temp = round(temp, 2)
           temp = str(temp)
           pressure = bme.values[1]
           humidity = bme.values[2]
           pressure = float(pressure)*0.03
           pressure = round(pressure, 2)
           pressure = str(pressure)
           rtc = RTC()
        
           now = rtc.datetime()
           now0 = str(now[0])
           now1 = str(now[1])
           now2 = str(now[2])
           now4 = str(now[4])
           now5 = str(now[5])
           now5 = int(now5)
           if now5 < 10:
             now5 = str(now5)
             now6 = str(0)
             now5 = (now6 + now5)
           elif now5 > 9:
             now5 = str(now5)
           print(now1,'-',now2,'-',now0)
           print(now4,':',now5)
           client = connection.accept()[0]
           request = client.recv(1024)
           request = str(request)       
           html = webpage(temp,pressure,humidity,now0,now1,now2,now4,now5)
           client.send(html)
           client.close()
           sleep(5)
        print("I'm going to sleep now.")
        deepsleep(60000)
        

try:
    
    ip = connect()
    connection = open_socket(ip)
    setTimeRTC()
    serve(connection)
except KeyboardInterrupt:
    machine.reset()

tenapier, 2023-10-14:
After trying to use it with a change made in the main.py program the connection was lost again, and no longer recognized when connected to the USB. It looks like it might be getting another location address, and maybe Thonny can't pick it up.
??????

tenapier, 2023-10-14:
Seems the PicoW came back to /dev/tty.usbmodem14601 after unplugging it and checking on the terminal with ls /dev/* and Thonny indicates it's at that location in the bottom right corner, but still not coming available under the Files category so I can't access it.

tenapier, 2023-10-14:
Only way to fix this so far is to nuke it, (flash_nuke.uf2), and reload the micropython uf2 file into. Unfortunately, anything stored in it previously is of course gone.

Planiska, 2023-10-19:
I have almost the same problem. Connecting to WIFI, collecting temperature etc. values, uploading values to MQTT server and then going to **deepsleep**. 
Is it possible that script starts when Pico is connected to the computer and goes to deepsleep and cannot accessed?

aivarannamaa, 2023-11-16:
> Is it possible that script starts when Pico is connected to the computer and Pico goes to deepsleep and cannot accessed?

Yes, this is very likely. Some people start their programs with something like `time.sleep(3)`, which creates a window of time when the program can be easily interrupted by tools like Thonny.

P-Kaempf, 2024-06-25:
You have an infinite loop and wonder why the controller does not respond? Put a break condition somewhere!

closed 2024-09-02 by aivarannamaa