thonny/thonny #2960: [bug] Trackpad and mouse not registering clicks in Thonny

dgnezda, 2023-10-13:
As the title says, Thonny has trouble registering my mouse clicks and trackpad clicks. I'm on an M1 MB Pro, running Sonoma. 

Mouse clicks don't get registered at all (no trouble everywhere else on mac), but trackpad clicks only work after I hard-press. Taps don't work. Again, elsewhere on mac and in other apps, everything works fine. I'm fine to use keyboard combos for most things, but sometimes you just need to click something.
I suspect this might be a mac OS Sonoma issue, as I don't recall having these issues before I updated my OS.
Hope this can get resolved in the future.
Cheers!

aivarannamaa, 2023-10-14:
Thanks for the report!

Unfortunately I can't reproduce this with M2 MB Pro, running Sonoma.

Which Thonny version are you using?

dgnezda, 2023-10-14:
I'm on Thonny 4.1.2, I'll update it to 4.1.3 and get back to you if the problem persists

dgnezda, 2023-10-14:
Yep, the problem is still present. So just to make it extra clear, tapping/clicking doesn't work on the main app window. It works when I tap in the main toolbar at the top (next to the apple logo .. File/Edit etc)

aivarannamaa, 2023-10-14:
Unfortunately I have no ideas how to investigate this further, as for me everything works with same Thonny version, same architecture and same OS version.

dgnezda, 2023-10-14:
Alright, I'l go the apple feedback route, see if they have any ideas. It might be that updating to Sonoma somehow broke it. I'll also try reinstalling Thonny, see if that fixes anything.

dgnezda, 2023-10-14:
Uninstalled the brew-installed version and tried the pip version, it's still the same.

vvnocode, 2023-10-22:
My MacBook is M1,and thonny's version 4.1.3. After I chosen interpret, this App didn't work.
[image]
[image]

Gussoh, 2023-10-29:
Same issue with Thonny 4.0.2 and Intel Mac running Sonoma is that is of any help.

aivarannamaa, 2023-10-30:
@vvnocode and @Gussoh, just to confirm -- are you both experience that trackpad taps and mouse clicks are not registered in Thonny, but trackpad hard presses are?

Here are my trackpad settings: 

[image]

Do you have anything different?

Gussoh, 2023-10-30:
Neither trackpad nor mouse presses are registering... maybe 50% of the time.

aivarannamaa, 2023-10-30:
> maybe 50% of the time.

@Gussoh, does it mean you need to click several times to get the effect or that at some days it works and some days it doesn't?

Gussoh, 2023-10-30:
It is on a per-click basis.
I just did a discovery when trying to replicate this. It has something to do with the mouse movement.

I just tried a few things:
Scenario 1:
1) Selected a section of text.
2) Moved the mouse to a different part of the text area
3) Keep mouse still and clicked 100 times, nothing happens
4) Moved the mouse a few pixels
5) Clicked the 101th time, cursor moved and selection as unselected.

Scenario 2:
1) Selected a section of text.
2) While moving the mouse, click once
3) Cursor moved

These two scenarios are deterministic for me.
One interpretation is that clicking doesn't work, but dragging does. 

I'd in general assume that this is not only in Thonny but in a lot of other apps too (perhaps especially the ones sharing the same UI framework or similar), but I have not noticed it in any other apps yet.

jskupien24, 2023-10-31:
I am experiencing the same issue on my MB pro with M2, running sonoma.

aivarannamaa, 2023-11-02:
A similar problem was posted at https://github.com/python/cpython/issues/110218 
I you can reproduce this in IDLE (Python IDE in the official distribution, e.g. https://www.python.org/ftp/python/3.12.0/python-3.12.0-macos11.pkg), then please let them know!

I still can't reproduce this myself, though.

dgnezda, 2023-11-02:
My case is exactly the same as that of @Gussoh. IDLE working fine on my end.

aivarannamaa, 2023-11-02:
@dgnezda, which version of IDLE did you try? Can you tell me its Tk version (reported in the About box)?

dgnezda, 2023-11-02:
I tried IDLE 3.11.5 with tcl/tk 8.6.12. **BUT**- now that I went into the menus, the buttons there were showing the same non-responsive behaviour. Earlier, I only tried the main menu dropdowns, where I could tap/click and everything was fine. The buttons in the about window (License / Copyright / Credits / Readme ...) are the ones where the behaviour of random non-responsiveness showed up again. 

So you think this might be a Tkinter issue?

aivarannamaa, 2023-11-02:
@dgnezda, with Tk 8.6.12 it is likely related to https://github.com/python/cpython/issues/110218, i.e. either tkinter or Tk issue. 

The puzzling part: Thonny 4.1 comes with Tk 8.6.13 (which can be checked from Thonny's About box) and according to posts at https://github.com/python/cpython/issues/110218, the problem is supposed to be fixed in Tk 8.6.13. I don't know whether what you're experiencing with Thonny 4.1 is another issue or there is something else in play (e.g. the bug still alive with the combination of Python 3.10 and Tk 8.6.13).

If I only could reproduce it myself ...

dgnezda, 2023-11-03:
I have installed Python 3.12.0 and IDLE works fine now. I tried reinstalling thonny and setting its interpreter to the latest Python version, but the issue is still there.
In case this helps in any way - the issue goes away once I reach OS-level menus. Like for example if I try to open a file in thonny, I need to hard-press the touch pad (normal taps don't work at all), then when I'm in the Finder-like window, taps work again.

dgnezda, 2023-11-05:
What versions of Python is everyone having this issue using?
For me, after installing Python 3.12.0 and restarting macOS a couple of times, the issue seems to have gone away completely.

aivarannamaa, 2023-11-05:
I had totally forgotten, that I have made an experimental bundle with Thonny 4.1.3 and Python 3.12: https://github.com/thonny/thonny/releases/download/v4.1.3/thonny-4.1.3-EXPERIMENTAL-py312.pkg

It would be nice if you could test this as well.

dgnezda, 2023-11-05:
Is there a way to have both versions installed? I'll gladly test it

eric-asu, 2023-11-06:
Same issue here, running M1 Pro and Sonoma

eric-asu, 2023-11-06:
I just installed the experimental version that @aivarannamaa shared and it is registering mouse clicks with no problem at all! Thanks!

G3zz, 2023-11-21:
I also get the issue on Sonoma 14.1.1 and Python 3.10.12

Oooklout, 2023-11-28:
I have the same problem. Everything was working fine doing PICO/Micropython development until I updated my MacBook Air M1 with Sonoma 14.1.1.  No other changes made to Thonny 4.12, Python 3.11.0 or anything else.
The new problem is that light-pressure clicks are ignored inside the Thonny window; pressing harder sometimes achieves a  normal click result.  Clicks on OS windows and other apps work fine. 

Footnote: Sadly I just tried to update Thonny with pip3 but then if when I select Thonny's Micropython Pico interpreter the Thonny app crashes. I used brew to install Thonny originally, 1 year ago, so maybe using pip3 this time was a bad idea!
I'll retry using brew.

joycepg, 2024-01-03:
same problem with thonny 4.1.4 installed via pip3 (on macOS Sonoma 14.2.1 (python is /Library/Frameworks/Python.framework/Versions/3.10).
standard clicks do not work, hard clicks sometimes work, only in thonny. 
also tried the experimental build above, and it seems to work ok.

joycepg, 2024-01-03:
i can also report latest python (3.12.1) from python.org, with a `pip3 install thonny` (4.1.4) produces a successful result.
after running `Install Certificates.command` the Manage Packages works fine too.

closed 2024-05-30 by aivarannamaa

mikahoy045, 2024-08-14:
I've experienced it, but in my experience my mac running low on batteries, when I plug the charger the mouse function completely in thony ide.