thonny/thonny #2961: Unable to flash ESP8266 from local file

pascalddg, 2023-10-14:
Using Thonny 4.1.3 to flash ESP8266 from a local file causes the ESP board unresponsive:

-  no REPL
-  receiving garbage over the COM interface. 
-  blue led is blinking rapidly

**Environment**:
- Thonny 4.1.3
- Windows 10
- NodeMCU and NodeMCU v3

Thonny 4.0.2 works well - I can flash the same image and get REPL via COM.

aivarannamaa, 2023-10-14:
Thonny 4.1.3 comes with newer version of esptool, which does the flashing. Perhaps something has changed there.

Please open the detailed installation options (Click on ☰ button and select "Show install options") and make sure all settings are as needed. You could try setting flash size to "detect" or flash mode to "dio".

codemee, 2023-11-21:
Setting flash size to "detect" **AND** flash mode to "dio" works.

closed 2023-11-21 by aivarannamaa