thonny/thonny #2963: esp32 can not connect, macos

cubehead, 2023-10-16:
PROBLEM IN THONNY'S BACK-END: Internal error (thonny.plugins.micropython.mp_back.ManagementError: Command output was not empty).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

I tested thonny, v 4.0.1 ~ v4.1.3 on macOS  14
same problem.
it may be the __thonny_helper class error
in windows, it works fine, but in macOS, all errors.

aivarannamaa, 2023-10-17:
Thanks for the report!

You could try this:

* Select "Tools => Open Thonny data folder"
* Reproduce the problem
* Send me the backend.log from the folder opened earlier.

cubehead, 2023-10-23:
14:51:14.392 ERROR   thonny.plugins.micropython.mp_back: Exception in MicroPython main method
Traceback (most recent call last):
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/micropython/mp_back.py", line 151, in __init__
    self._prepare_after_soft_reboot(clean)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/micropython/mp_back.py", line 173, in _prepare_after_soft_reboot
    self._execute_without_output(script)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/micropython/mp_back.py", line 553, in _execute_without_output
    raise ManagementError("Command output was not empty", script, out, err)
thonny.plugins.micropython.mp_back.ManagementError: Command output was not empty
14:51:14.395 ERROR   thonny.plugins.micropython.mp_back: ManagementError details:
SCRIPT: class __thonny_helper:
    import builtins
    try:
        import uos as os
    except builtins.ImportError:
        import os
    import sys
    last_non_none_repl_value = None

    # for object inspector
    inspector_values = builtins.dict()
    @builtins.classmethod
    def print_repl_value(cls, obj):
        if obj is not None:
            cls.builtins.print('[object_link_for_thonny=%d]' % cls.builtins.id(obj), cls.builtins.repr(obj), '[/object_link_for_thonny]', sep='')
            cls.last_non_none_repl_value = obj

    @builtins.classmethod
    def print_mgmt_value(cls, obj):
        cls.builtins.print('<thonny>', cls.builtins.repr(obj), '</thonny>', sep='', end='')

    @builtins.classmethod
    def repr(cls, obj):
        try:
            s = cls.builtins.repr(obj)
            if cls.builtins.len(s) > 50:
                s = s[:50] + "..."
            return s
        except cls.builtins.Exception as e:
            return "<could not serialize: " + __thonny_helper.builtins.str(e) + ">"

    @builtins.classmethod
    def listdir(cls, x):
        if cls.builtins.hasattr(cls.os, "listdir"):
            return cls.os.listdir(x)
        else:
            return [rec[0] for rec in cls.os.ilistdir(x) if rec[0] not in ('.', '..')]

    @builtins.classmethod
    def getcwd(cls):
        return cls.os.getcwd()

    @builtins.classmethod
    def chdir(cls, x):
        return cls.os.chdir(x)

    @builtins.classmethod
    def rmdir(cls, x):
        return cls.os.rmdir(x)

STDOUT: 

STDERR: Traceback (most recent call last):
  File "<stdin>", line 15
SyntaxError: invalid syntax

Warths, 2023-11-19:
Hello. I'm encountering the same issue. I must say that this happens when i'm using a print statement on an hardware timer callback. It works fine after what looks like a random amount of resets. I'm using an ESP32 on Thonny 4.1.3, MacOS 14.0

`
19:00:55.338 [MainThread] INFO    thonny: TIME/MODS 23.888 got welcome 
19:00:55.339 [MainThread] INFO    thonny: TIME/MODS 0.001 bef preparing helpers 
19:00:55.339 [MainThread] INFO    thonny.plugins.micropython.mp_back: Preparing helpers
19:00:55.340 [MainThread] INFO    thonny: TIME/MODS 0.001 befsubcode 
19:00:55.340 [MainThread] INFO    thonny.plugins.micropython.bare_metal_backend: Requesting raw mode at b'>>> '
19:00:55.345 [MainThread] INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: '\r\n'
19:00:55.357 [MainThread] INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: ''
19:00:55.357 [MainThread] INFO    thonny.plugins.micropython.bare_metal_backend: Entered raw prompt
19:00:55.430 [MainThread] INFO    thonny: TIME/MODS 0.090 affsubcode 
19:00:55.603 [MainThread] INFO    thonny: TIME/MODS 0.173 affforw 
19:00:55.603 [MainThread] INFO    thonny: TIME/MODS 0.000 prepared helpers 
19:00:55.603 [MainThread] INFO    thonny: TIME/MODS 0.000 befsubcode 
19:00:55.617 [MainThread] INFO    thonny: TIME/MODS 0.014 affsubcode 
19:00:55.632 [MainThread] INFO    thonny: TIME/MODS 0.015 affforw 
19:00:55.632 [MainThread] INFO    thonny: TIME/MODS 0.000 got cwd 
19:00:55.632 [MainThread] INFO    thonny: TIME/MODS 0.000 befsubcode 
19:00:55.647 [MainThread] INFO    thonny: TIME/MODS 0.015 affsubcode 
19:00:55.662 [MainThread] INFO    thonny: TIME/MODS 0.015 affforw 
19:00:55.662 [MainThread] INFO    thonny: TIME/MODS 0.000 prepared 
19:00:55.662 [MainThread] INFO    thonny.plugins.micropython.mp_back: Prepared
19:00:55.662 [MainThread] INFO    thonny: TIME/MODS 0.000 befsubcode 
19:00:55.677 [MainThread] INFO    thonny: TIME/MODS 0.015 affsubcode 
19:00:55.739 [MainThread] INFO    thonny: TIME/MODS 0.062 affforw 
19:00:55.739 [MainThread] INFO    thonny: TIME/MODS 0.000 befsubcode 
19:00:55.762 [MainThread] INFO    thonny: TIME/MODS 0.023 affsubcode 
19:00:55.830 [MainThread] INFO    thonny: TIME/MODS 0.068 affforw 
19:00:55.831 [MainThread] INFO    thonny: TIME/MODS 0.000 befsubcode 
19:00:55.858 [MainThread] INFO    thonny: TIME/MODS 0.027 affsubcode 
19:00:55.926 [MainThread] INFO    thonny: TIME/MODS 0.068 affforw 
19:00:55.926 [MainThread] INFO    thonny.backend: Loading Jedi
19:00:55.926 [MainThread] INFO    thonny: TIME/MODS 0.000 Before loading Jedi 
19:00:55.926 [MainThread] INFO    thonny.common: Using THONNY_FRONTEND_SYS_PATH ['/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python310.zip', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/lib-dynload', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages', '/Users/warths/Library/Thonny/plugins/lib/python/site-packages']
19:00:55.970 [MainThread] INFO    thonny: TIME/MODS 0.043 After loading Jedi (+121 modules)
19:00:55.970 [MainThread] INFO    thonny: NEW MODS ['_blake2', '_compat_pickle', '_hashlib', '_json', '_osx_support', '_pickle', '_sysconfigdata__darwin_darwin', 'difflib', 'filecmp', 'gc', 'hashlib', 'importlib._abc', 'importlib.util', 'jedi', 'jedi._compatibility', 'jedi.api', 'jedi.api.classes', 'jedi.api.completion', 'jedi.api.completion_cache', 'jedi.api.environment', 'jedi.api.errors', 'jedi.api.exceptions', 'jedi.api.file_name', 'jedi.api.helpers', 'jedi.api.interpreter', 'jedi.api.keywords', 'jedi.api.project', 'jedi.api.refactoring', 'jedi.api.refactoring.extract', 'jedi.api.strings', 'jedi.cache', 'jedi.common', 'jedi.debug', 'jedi.file_io', 'jedi.inference', 'jedi.inference.analysis', 'jedi.inference.arguments', 'jedi.inference.base_value', 'jedi.inference.cache', 'jedi.inference.compiled', 'jedi.inference.compiled.access', 'jedi.inference.compiled.getattr_static', 'jedi.inference.compiled.mixed', 'jedi.inference.compiled.subprocess', 'jedi.inference.compiled.subprocess.functions', 'jedi.inference.compiled.value', 'jedi.inference.context', 'jedi.inference.docstring_utils', 'jedi.inference.docstrings', 'jedi.inference.filters', 'jedi.inference.flow_analysis', 'jedi.inference.gradual', 'jedi.inference.gradual.annotation', 'jedi.inference.gradual.base', 'jedi.inference.gradual.conversion', 'jedi.inference.gradual.generics', 'jedi.inference.gradual.stub_value', 'jedi.inference.gradual.type_var', 'jedi.inference.gradual.typeshed', 'jedi.inference.gradual.typing', 'jedi.inference.gradual.utils', 'jedi.inference.helpers', 'jedi.inference.imports', 'jedi.inference.lazy_value', 'jedi.inference.names', 'jedi.inference.param', 'jedi.inference.parser_cache', 'jedi.inference.recursion', 'jedi.inference.references', 'jedi.inference.signature', 'jedi.inference.syntax_tree', 'jedi.inference.sys_path', 'jedi.inference.utils', 'jedi.inference.value', 'jedi.inference.value.decorator', 'jedi.inference.value.dynamic_arrays', 'jedi.inference.value.function', 'jedi.inference.value.instance', 'jedi.inference.value.iterable', 'jedi.inference.value.klass', 'jedi.inference.value.module', 'jedi.parser_utils', 'jedi.plugins', 'jedi.plugins.django', 'jedi.plugins.flask', 'jedi.plugins.pytest', 'jedi.plugins.registry', 'jedi.plugins.stdlib', 'jedi.settin [... cut ...]
        if cls.builtins.hasattr(cls.os, "listdir"):
            return cls.os.listdir(x)
        else:
            return [rec[0] for rec in cls.os.ilistdir(x) if rec[0] not in ('.', '..')]

    @builtins.classmethod
    def getcwd(cls):
        return cls.os.getcwd()

    @builtins.classmethod
    def chdir(cls, x):
        return cls.os.chdir(x)

    @builtins.classmethod
    def rmdir(cls, x):
        return cls.os.rmdir(x)

STDOUT: (0, 0, 0)
(0, 0, 0)

STDERR: 

19:56:04.789 [MainThread] INFO    thonny.backend: After mainloop
`

NicK4rT, 2024-10-04:
I'm running into the same issue, also using an ESP32 and I too first encountered it at a random print statement, although I can't seem to find just one specific line that causes it now. The issue persisted consistently though. And it also seems to depend on the device, for the same code I encounter it on one ESP32 but not with another ESP32. 

I also can't seem to figure out anymore where it occurs, as for some reason enabling my debug messages fixes the issue. ?!?!?!

So far I figured out, that the code doesn't break if I add a sleep statement at the end. I feel that there is something going on in the background, which if it is given enough time it works itself out, but otherwise it breaks (also why I think the debug statements make it work, as these take more time to run). Specifically the required time for it to not break is around 0.01s (my code has some variability in how long it takes to run, so it might be 0.011s or 0.009s).

`14:23:11.210 [MainThread] INFO    thonny: Thonny version: 4.1.6
14:23:11.210 [MainThread] INFO    thonny: cwd: /Users/nickart
14:23:11.210 [MainThread] INFO    thonny: original argv: ['/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/Resources/Python.app/Contents/MacOS/Python', '-s', '-u', '-B', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/esp/esp_back.py', "{'clean': True, 'port': '/dev/cu.usbmodem101', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': True, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'ESP32Proxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
14:23:11.210 [MainThread] INFO    thonny: sys.executable: /Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/bin/python3.10
14:23:11.210 [MainThread] INFO    thonny: sys.argv: ['/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/esp/esp_back.py', "{'clean': True, 'port': '/dev/cu.usbmodem101', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': True, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'ESP32Proxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
14:23:11.210 [MainThread] INFO    thonny: sys.path: ['/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/esp', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python310.zip', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/lib-dynload', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages']
14:23:11.210 [MainThread] INFO    thonny: sys.flags: sys.flags(debug=0, inspect=0, interactive=0, optimize=0, dont_write_bytecode=1, no_user_site=1, no_site=0, ignore_environment=0, verbose=0, bytes_warning=0, quiet=0, hash_randomization=1, isolated=0, dev_mode=False, utf8_mode=0, warn_default_encoding=0, int_max_str_digits=-1)
14:23:11.210 [MainThread] INFO    thonny.plugins.micropython.bare_metal_backend: Starting backend, args: {'clean': True, 'port': '/dev/cu.usbmodem101', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': True, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'ESP32Proxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}
14:23:11.228 [MainThread] INFO    thonny.plugins.micropython.bare_metal_backend: Initial submit_mode: raw_paste, write_block_size: 127, write_block_delay: 0, read_block_size: 0
14:23:11.228 [MainThread] INFO    thonny.plugins.micropython.mp_back: Initializing MicroPythonBackend of type EspMicroPythonBackend
14:23:11.228 [MainThread] INFO    thonny: TIME/MODS 0.117 before prepare (+73 modules)
14:23:11.228 [MainThread] INFO    thonny: NEW MODS ['__future__', '_bz2', '_compression', '_datetime', '_heapq', '_lzma', '_posixsubprocess', '_queue', '_struct', '_tkinter', 'array', 'binascii', 'bz2', 'datetime', 'encodings.ascii', 'errno', 'faulthandler', 'fcntl', 'fnmatch', 'gettext', 'heapq', 'lzma', 'math', 'ntpath', 'pathlib', 'platform', 'queue', 'select', 'selectors', 'serial', 'serial.serialposix', 'serial.serialutil', 'shlex', 'shutil', 'signal', 'struct', 'subprocess', 'termios', 'textwrap', 'thonny.backend', 'thonny.base_file_browser', 'thonny.codeview', 'thonny.config_ui', 'thonny.editors', 'thonny.languages', 'thonny.misc_utils', 'thonny.plugins', 'thonny.plugins.backend_config_page', 'thonny.plugins.micropython', 'thonny.plugins.micropython.bare_metal_backend', 'thonny.plugins.micropython.connection', 'thonny.plugins.micropython.mp_back', 'thonny.plugins.micropython.mp_common', 'thonny.plugins.micropython.mp_front', 'thonny.plugins.micropython.serial_connection', 'thonny.plugins.micropython.webrepl_connection', 'thonny.roughparse', 'thonny.running', 'thonny.tktextext', 'thonny.ui_utils', 'thonny.workdlg', 'tkinter', 'tkinter.commondialog', 'tkinter.constants', 'tkinter.dialog', 'tkinter.filedialog', 'tkinter.font', 'tkinter.messagebox', 'tkinter.simpledialog', 'tkinter.ttk', 'urllib', 'urllib.parse', 'zlib']
14:23:11.229 [MainThread] INFO    thonny.plugins.micropython.mp_back: Sending interrupt
[... cut ...]
14:23:11.746 [MainThread] INFO    thonny: TIME/MODS 0.000 sent ready 
14:23:11.746 [MainThread] INFO    thonny: TIME/MODS 0.000 Beginning of mainloop 
14:23:11.746 [MainThread] INFO    thonny: TIME/MODS 0.000 before get_globals 
14:23:11.746 [MainThread] INFO    thonny: TIME/MODS 0.000 befsubcode 
14:23:11.749 [MainThread] INFO    thonny: TIME/MODS 0.003 affsubcode 
14:23:11.778 [MainThread] INFO    thonny: TIME/MODS 0.028 affforw 
14:23:11.778 [MainThread] INFO    thonny: TIME/MODS 0.000 after get_globals 
`