thonny/thonny #2967: Support Palestine kids

cornerot, 2023-10-19:
Can you add the Palestine flag to the menu bar next to the Ukrainian one?

aivarannamaa, 2023-10-19:
No, I'll keep my focus on what's happening in Ukraine. Parts of Ukraine are still occupied by Russia and I won't support the attempts to exploit the situation in Near East and draw world's attention away from this.

closed 2023-10-19 by aivarannamaa

webdeb, 2024-02-23:
This is actually a very good idea. I think we should ask the micropython community to bring some more engagement into the political aspect of Thonny. There are a lot of people who will love to use Thonny as a platform for their views.

rosesonfire, 2024-12-22:
Free free Palestine! End Zionist occupation of Palestine! IDF is a terrorist organization. SPEAK UP FOR THE FREEDOM OF PALESTINIANS FROM ZIONISTS.