thonny/thonny #2968: PyTorch not installing in Thonny

rwblues, 2023-10-19:
[PyTorch Error.txt](https://github.com/thonny/thonny/files/13047771/PyTorch.Error.txt)

This happens with packages and plug-ins.

Is there somthing I can do about it? Do you need more information?

Kind regards

aivarannamaa, 2023-10-20:
Did you read the error you sent to me? ;)

I don't now anything about PyTorch, but this error output is pretty explicit about the problem. I know that the error messages can be intimidating, and the useful sentence can be buried between heaps of gibberish, but I suggest to read these anyway.

Let me know if you still need help with this!

cnoblejr, 2025-01-01:
Same here cannot install pytorch.  Guess I'll have to use Jupyter Lab.

aivarannamaa, 2025-01-15:
@cnoblejr, in case you got the same error, you should read the suggestion given in the error text.

closed 2025-01-15 by aivarannamaa