thonny/thonny #2969: Thonny installed with flatpak does not import the turtle module

zeyalex, 2023-10-20:
Thonny installed with flatpak does not import the turtle module. Installed from system packages, the module was imported without problems. But there is version 3.3.14.
My System:
  Kernel: 5.15.0-87-generic x86_64 bits: 64 compiler: gcc v: 11.4.0 Desktop: Xfce 4.18.1
    tk: Gtk 3.24.33 wm: xfwm dm: LightDM Distro: Linux Mint 21.2 Victoria base: Ubuntu 22.04 jammy

aivarannamaa, 2023-11-16:
For flatpak I recommend creating an issue here: https://github.com/flathub/org.thonny.Thonny (@jwillikers may have something to say about this).

jwillikers, 2023-11-16:
@zeyalex You're attempting to use the built-in `turtle` Python module, right? It's explicitly deleted from the Python in the freedesktop Flatpak [here](https://gitlab.com/freedesktop-sdk/freedesktop-sdk/-/blob/master/elements/components/python3.bst#L44-45). I should be able to add it back in the Thonny Flatpak itself.

jwillikers, 2023-11-16:
This has been fixed in flathub/org.thonny.Thonny#233. The fix should be available in a few hours time.

closed 2023-11-18 by aivarannamaa