thonny/thonny #2974: First class support for Python virtual environments

Gadgetoid, 2023-10-23:
Visual Studio Code's Python extension understands the `~/.virtualenvs' (Virtualenvwrapper) directory as a standard location for virtual environments. It would be great if Thonny could support this, since it permits users to `workon project` a project created within the scope of Thonny.

Broadly I think Thonny should search standard paths for virtual environment collections and give them first class support in the "Configure interpreter" dialog. IE:

1. Showing "Virtual Environments in ~/.virtualenvs" in the top drop-down and,
2. A list of environments within `~/.virtualenvs' in the one below
3. Some means of adding a new one, using venv/virtualenv/virtualenvwrapper/conda etc depending on context

Here's where VSCode searches: https://code.visualstudio.com/docs/python/environments#_where-the-extension-looks-for-environments

This expands upon and relates to issue #2096

aivarannamaa, 2023-10-30:
Thanks for the suggestion and PR!

I haven't had time to form my opinion on the topic discussed under https://github.com/raspberrypi/bookworm-feedback/issues/100, but it is interesting topic for me and I try to work through it this week.

aivarannamaa, 2024-06-09:
I've implemented a dialog for creating a virtual environments and added a shortcut for it into the interpreter menu.

[image]

[image]

closed 2024-06-09 by aivarannamaa

chuckf201, 2024-09-08:
Thonny and Bookworm are not compatable for new users. The non-venv world was easier to manage.
All I wantd to do was import bme680 library into my code and run it!.
I seems to be rocket-science to use Bookworm.
I've been playing with this for many days and totally frustrated.
I'll wait for a fix. Yes, it's broken.

Gadgetoid, 2024-09-09:
Thonny could- at best- opt for using a venv by default but this would not match the experience an intermediate (or advanced) user would get from RPi OS’s command line.

A proper fix to this problem is out of Thonny’s hands and would require RPi to:

* add user scripts to create and activate a venv by default
* Change the Thonny shortcut to use this venv

I don’t get the impression that they’re interested in doing this and since the default gpio library (lg) afaik still cannot be installed into a virtualenv (despite my best efforts) they would run into trouble anyway.

I’m not sure if the Thonny shipped with Bookworm has caught up with the latest little fixes upstream here, but I should probably check!

But yeah TLDR: I agree with you wholly and have been trying to remedy this.

Gadgetoid, 2024-09-09:
Changing the `LocalCPython` section of `~/.config/Thonny/configuration.ini` allowed me to launch Thonny on Bookworm, targeting the "pimoroni" virtual env (set up automagically when using any of our new style installers):

```
[LocalCPython]
last_configurations = [{'run.backend_name': 'LocalCPython', 'LocalCPython.executable': '/home/pi/.virtualenvs/pimoroni/bin/python'}]
executable = /home/p[/.virtualenvs/pimoroni/bin/python
```

I couldn't find any other way to specify this as a launch option, but you can also make this change via Thonny's configuration UI (accessible via the little three bars in the bottom right corner of the application window).

If you use virtualenvwrapper you can type `workon <virtualenvname>` to switch quickly in CLI.

`sudo apt install python3-virtualenvwrapper virtualenvwrapper` then `source /usr/share/virtualenvwrapper/virtualenvwrapper.sh` (helps to add the latter to your bash profile.)

I started writing a guide to *some* of this stuff here but it's not well publicised: https://pimoroni.github.io/venv-python/