thonny/thonny #2975: Thonny not connecting to Pico

W-Michael, 2023-10-23:
My OS is: Linux Mint 20.2
Kernel : Linux 5.4.0-91-generic (x86_64)
Version : #102-Ubuntu SMP Fri Nov 5 16:31:28 UTC 2021

Raspberry PiPico is connected, lsusb command output is: Bus 006 Device 002: ID 2e8a:0005 MicroPython Board in FS mode

With Thonny installed as a snap startup produces a "Error on initializing backend" popup containing:

Traceback (most recent call last):
File "/snap/thonny/210/lib/python3.10/site-packages/thonny/workbench.py", line 524, in _start_runner
self._runner.start()
File "/snap/thonny/210/lib/python3.10/site-packages/thonny/running.py", line 155, in start
self.restart_backend(False, True)
File "/snap/thonny/210/lib/python3.10/site-packages/thonny/running.py", line 731, in restart_backend
self._proxy = backend_class(clean)
File "/snap/thonny/210/lib/python3.10/site-packages/thonny/plugins/micropython/mp_front.py", line 91, in __init__
self._fix_port()
File "/snap/thonny/210/lib/python3.10/site-packages/thonny/plugins/micropython/mp_front.py", line 114, in _fix_port
potential = self._detect_potential_ports()
File "/snap/thonny/210/lib/python3.10/site-packages/thonny/plugins/micropython/mp_front.py", line 199, in _detect_potential_ports
all_ports = list_serial_ports()
File "/snap/thonny/210/lib/python3.10/site-packages/thonny/plugins/micropython/mp_front.py", line 989, in list_serial_ports
_PORTS_CACHE = _list_serial_ports_uncached(skip_logging=skip_logging)
File "/snap/thonny/210/lib/python3.10/site-packages/thonny/plugins/micropython/mp_front.py", line 1023, in _list_serial_ports_uncached
for p in comports():
File "/snap/thonny/210/lib/python3.10/site-packages/serial/tools/list_ports_linux.py", line 102, in comports
for info in [SysFS(d) for d in devices]
File "/snap/thonny/210/lib/python3.10/site-packages/serial/tools/list_ports_linux.py", line 102, in <listcomp>
for info in [SysFS(d) for d in devices]
File "/snap/thonny/210/lib/python3.10/site-packages/serial/tools/list_ports_linux.py", line 48, in __init__
num_if = int(self.read_line(self.usb_device_path, 'bNumInterfaces'))
TypeError: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'

All the script related options in the "Run" tab are greyed out

This is my first entry into the world of Thonny, HELP

Thanks
Michael

capecrusader-121, 2023-10-24:
Can you install and run snappy-debug.

In terminal;
```
sudo snap install snappy-debug
snappy-debug
snap run thonny
**Execute the work**
```

The outputs of snappy-debug will be helpful to say the least.

W-Michael, 2023-10-24:
Hi Sameer
I am not sure I followed your instructions correctly but here is the output
$ snappy-debug
INFO: Following '/var/log/syslog'. If have dropped messages, use:
INFO: $ sudo journalctl --output=short --follow --all | sudo snappy-debug
snap run thonny
= AppArmor =
Time: Oct 24 15:30:56
Log: apparmor="DENIED" operation="open" profile="snap.zotero-snap.zotero-snap" name="/proc/219811/net/arp" pid=219811 comm=4C696E6B204D6F6E69746F72 requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
File: /proc/219811/net/arp (read)
Suggestions:
* adjust program to not access '@{PROC}/@{pid}/net/arp'
* add one of 'firewall-control, network-control, network-observe' to 'plugs'

**Execute the work**

Sent with [Proton Mail](https://proton.me/) secure email.

W-Michael, 2023-10-24:
Hi Sameer
I looked at the code where the error occurs and the line in in a try:/except: which should handle the error and not lead to a traceback. That is in my limited understanding of python.
Cheers

Michael

Sent with [Proton Mail](https://proton.me/) secure email.

capecrusader-121, 2023-10-25:
@W-Michael  Yes, the snap has some outstanding issues with  Raspberry PiPico in the past too, my aim was to use snappy-debug to tell weather the snap as a confined app lacks permission to any interface when that device is being tried to use ?
Is this the full output of snappy-debug ?

P.S. try the following :

```
sudo snap connect thonny:removable-media :removable-media
sudo snap connect thonny:raw-usb :raw-usb
```

Note execute the above while the snap isn't running.

Hope this helps, if not respond back with then new results.

W-Michael, 2023-10-25:
Hi Sameer
Success !!
Followed your instructions then Thonny produced:
Try adding yourself to the 'dialout' group:
> sudo usermod -a -G dialout <username>
(NB! You may need to reboot your system after this!)
Process ended with exit code 1.
Followed those instructions and now I am connecting to the Pico and can start learning to work with it.
Thanks !!!!

Michael

Sent with [Proton Mail](https://proton.me/) secure email.

capecrusader-121, 2023-10-25:
@W-Michael  So i believe the solution to the pico trouble is lack of access to raw-usb interface by the snap, i will consider seeking autoconnect of this interface in the future to ensure the best results for the snap users.

capecrusader-121, 2023-10-25:
@W-Michael  If this is fixed consider closing this.

W-Michael, 2023-10-25:
Hi Sameer
Your help was excellent and much appreciated!!
However It took time, effort, generous help from others and luck to make contact with you.
A note on the Tonny website alerting users to this issue and instructions on the workaround would be quite useful.

Cheers
Michael

Sent with [Proton Mail](https://proton.me/) secure email.

capecrusader-121, 2023-11-27:
Hey there @W-Michael  @aivarannamaa  , happy to inform this issue has been fixed by seeking auto-connect request for raw-usb interface, that is pico devices and others will now not need manual work to get the snap working with such devices, see --> https://forum.snapcraft.io/t/autoconnect-request-for-thonny-1/37797/4
 and https://bugs.launchpad.net/thonny-snap/+bug/2040520 for more info, _**this issue is resolved and hence can be closed.**_

aivarannamaa, 2023-11-27:
Thanks!
@W-Michael, please close this if you can confirm the problem is solved.