thonny/thonny #2976: error message

eugenerod33, 2023-10-24:
Traceback (most recent call last):
  File "/usr/lib/python3.8/tkinter/__init__.py", line 1892, in __call__
    return self.func(*args)
  File "/usr/lib/python3/dist-packages/thonny/shell.py", line 1268, in _clear_shell
    self._clear_content(end_index)
  File "/usr/lib/python3/dist-packages/thonny/shell.py", line 1361, in _clear_content
    btn_pos = float(self.index(btn))
ValueError: could not convert string to float: ''

Moosems, 2023-10-25:
Could you give more information on reproducing this issue?

aivarannamaa, 2023-11-16:
@eugenerod33, if you still have this problem with the latest Thonny version (4.1.3) then please reopen this issue and try to find the steps for reproducing this.

closed 2023-11-16 by aivarannamaa