thonny/thonny #2979: CANT INSTALL PACKAGES TO PICO USING THONNY

doingus, 2023-10-24:
install --progress-bar off micropython-servo
PROBLEM IN THONNY'S BACK-END: Exception while handling 'install_distributions' (AssertionError).
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

See Thonny's backend.log for more info.

Process ended with exit code 1.
 I get this error everytime i try to install a package
 ive restarted multiple times

doingus, 2023-10-24:
Traceback (most recent call last):
  File "C:\Users\HAHA\AppData\Local\Programs\Thonny\lib\tkinter\__init__.py", line 1921, in __call__
    return self.func(*args)
  File "C:\Users\HAHA\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\pip_gui.py", line 331, in _on_install_click
    self._perform_pip_action("install")
  File "C:\Users\HAHA\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\pip_gui.py", line 772, in _perform_pip_action
    if self._has_remote_target():
  File "C:\Users\HAHA\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\pip_gui.py", line 1006, in _has_remote_target
    return get_runner().get_backend_proxy().supports_remote_files()
AttributeError: 'NoneType' object has no attribute 'supports_remote_files'

Sherzodbek1426, 2023-10-25:
Traceback (most recent call last):
File "C:\Users\HAHA\AppData\Local\Programs\Thonny\lib\tkinter_init_.py", line 1921, in call
return self.func(*args)
File "C:\Users\HAHA\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\pip_gui.py", line 331, in _on_install_click
self._perform_pip_action("install")
File "C:\Users\HAHA\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\pip_gui.py", line 772, in _perform_pip_action
if self._has_remote_target():
File "C:\Users\HAHA\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\pip_gui.py", line 1006, in _has_remote_target
return get_runner().get_backend_proxy().supports_remote_files()
AttributeError: 'NoneType' object has no attribute 'supports_remote_files'

illustratordavid, 2023-10-26:
Same issue for me in 4.1.3, downgrading to 4.1.2 installs same library fine

doingus, 2023-10-27:
> Same issue for me in 4.1.3, downgrading to 4.1.2 installs same library fine

This worked for me as well thank you!

jansierakowski, 2023-11-04:
For me 4.0.2 only works. Pico W with v1.21 uf2

aivarannamaa, 2023-11-18:
Thank you for the report and sorry for the late reply! I could reproduce this and will fix it soon in version 4.1.4.

closed 2023-11-18 by aivarannamaa

aivarannamaa, 2023-11-19:
Please try https://github.com/thonny/thonny/releases/tag/v4.1.4