thonny/thonny #2983: Quite wierd bug while using the serial

mirkokral, 2023-10-28:
(I checked, it does not happen in PuTTY. The python code just errors like normal and no duplicate errors.)
Board: micro-bit
Firmware: Adafruit CircuitPython 8.2.6 (Installed through thonny's built in installer)

Console:
```
>>> aa aaaaaaaa aaaaaaaa
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
>OKTraceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'aaaaaa' is not defined
>>> a
Serial console setup

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
```

How to reproduce:
Run a long string of stuff e. "aa aaaaaaaa aaaaaaaa" (invalid stuff, if you see 2 different errors, and a wierd char with >OK then you have succeeded)
Run a short string e. "a"
It should crash with a hard fault and a memory access or instruction error.

Console:
```
>>> aa aaaaaaaa aaaaaaaa
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
>OKTraceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'aaaaaa' is not defined
>>> a
Serial console setup

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
```

As i am new in circuitpython, i don't really know what is happening and why is it happening.

aivarannamaa, 2023-11-16:
Thank you for the report and sorry for the late reply!

Which OS and which Thonny version are you using? Can you reproduce this with CP 8.2.7? I cannot in macOS.

mirkokral, 2024-04-14:
> Thank you for the report and sorry for the late reply!
> 
> Which OS and which Thonny version are you using? Can you reproduce this with CP 8.2.7? I cannot in macOS.

Sorry for very late reply but i was using arch linux. Forgot the thonny version.

aivarannamaa, 2024-07-12:
> CircuitPython core code crashed hard

I'm afraid this is out of my reach. Most likely this is fixed in newer CircuitPython versions

closed 2024-07-12 by aivarannamaa