thonny/thonny #2984: Executing badly formatted commands in shell causes program lock up

zombieregime, 2023-10-31:
Stil kind of newish to the whole git thing, didnt seem like this has been posted before. Could be a general python (3.10.8, on debian bullseye) issue, but it happened in Thonny so reporting it here in case mitigations in Thonny could be integrated, if feasible.

Was working on a unsaved script (yes, I know...) loaded via the debug gear icon in the shell pane. Had a for loop from some hammering away at syntax beforehand...

`python >>> for i in range(5):`
`              flop = i`
`              print(flip(flop))`

...Pressed up to bring command back up, edited the flop variable, deleted the for line, and on an instinctual finger twitch hit enter, ie...

`python >>> _`
 `               flop = 1`
`               print(flip(flop))`

...causing the UI of Thonny to become unresponsive, and the PC speaker to start chirping like it was trying to do error bell on top of error bell. CPU load pegged for the core (I assume) it was on. Opened task manager and stopped the process (literally right click, and stop, the process froze as expected), the system did a normal full error beep as Thonny does, UI unresponsive as expected. Gave the process low priority, and "continued" it. Went into Thonny and clicked like mad on the stop sign hoping to get an instruction in to stop whatever was happening on the backend. Could not get it through, tried control+c a few times, eventually I just killed the Thonny process losing the script into the wild blue yander. 

Hopefully all that formatting works. Like I said, could be something Python does in shell too, haven't tried it in the python shell from a terminal window outside of Thonny. But was in Thonny when it happened so here we are.

Otherwise I enjoy the IDE(?) coming from coding python scripts in nano and notepad++. Great work to all involved.

aivarannamaa, 2023-11-16:
Thank you for the report and sorry for the late reply!

I could reproduce this and will fix it soon.

zombieregime, 2023-11-16:
No worries. Regardless of how long a software has been in a package tree, or not, I view the whole open source development scene as at the grace, leisure, and will of the developers (or the community whichever is greater, whatever a given case may be), and am just thankful for the efforts various devs (and dev groups) have put into the various projects out there. Im Glad I could offer something towards bettering the software on some level (even if just avoiding an odd lock up scenario). As mentioned, on the whole very much enjoying Thonny (the whole stack inspection side bar thing is an amazing view into how the code is executing, especially for us a bit new to the deeper goings on under the interpreters hood), another internet high five to all involved. Thanks for the time and efforts put into this project, much appreciated!

aivarannamaa, 2023-11-18:
Thanks for the kind words!