thonny/thonny #2989: unable to flash pico on thonny

Shahzebkhan762, 2023-11-02:
Could not download variants info from https://raw.githubusercontent.com/thonny/thonny/master/data/micropython-variants-uf2.json

this is what i am getting when i am trying to connect with pico and run
[image]

Shahzebkhan762, 2023-11-02:
[image]
also my device manager shows there are issues with driver when i connect pico with my CPU

KubaO, 2023-11-04:
There's not much Thonny can help with here. Look at the RPi Pico documentation to see how to force a boot into the bootloader. Since the bootloader is in the ROM in the RP2040 chip, there's no way to brick it. It should show up as a storage device. The Pico needs to be removed from the target board to be sure GPIO and power connections don't mess things up. Hope that helps.

aivarannamaa, 2023-11-16:
@Shahzebkhan762, can you access https://raw.githubusercontent.com/thonny/thonny/master/data/micropython-variants-uf2.json via your browser?

Shahzebkhan762, 2023-11-17:
> @Shahzebkhan762, can you access https://raw.githubusercontent.com/thonny/thonny/master/data/micropython-variants-uf2.json via your browser?

Yes I can

Shahzebkhan762, 2023-11-17:
@aivarannamaa Yes i can