thonny/thonny #2991: install thonny

woofwoof, 2023-11-03:
For some reason I couldn't install Thonny easily (as a complete package) under Win 11 but eventually installed python from python.org and separately installed thonny using "pip3.exe install thonny" then created a shortcut - all documented on the wiki with the exception of the quoted command string that needs ammending ever so slightly.

aivarannamaa, 2023-11-16:
Thank you for the report!

> For some reason I couldn't install Thonny easily (as a complete package) under Win 11

Did you get an error message?

woofwoof, 2023-11-19:
No problem but unfortunately I didn't note an error message. It worked, as stated, although using that circuitous method.

aivarannamaa, 2023-11-24:
> unfortunately I didn't note an error message.

In this case I'm afraid I can't do anything.

closed 2023-11-24 by aivarannamaa