thonny/thonny #2992: error message

eugenerod33, 2023-11-04:
I was working on Euler project problem 8 in the terminal window of Thonny.  Here's a copy of my input and the response:

>>> num = 73167176531330624919225119674426574742355349194934
96983520312774506326239578318016984801869478851843
PROBLEM WITH THONNY'S BACK-END:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thonny/backend.py", line 1245, in execute_source
    statements = compile(ast.Module(body=root.body[:-1]), filename, "exec")
TypeError: required field "type_ignores" missing from Module

aivarannamaa, 2023-11-16:
Thank you for the report and sorry for the late reply!

Which Thonny version were you using?

eugenerod33, 2023-11-17:
4.1.3

aivarannamaa, 2023-11-18:
This doesn't look right... Thonny 4.1.3 does not contain this piece of code, which is mentioned in the stacktrace.

Could you send me the Thonny's `backend.log`? It can be found be selecting "Tools => Open Thonny data folder".

eugenerod33, 2023-11-19:
Sorry, but I don't know how to do that.  When I open the log with a text editor, I get a blank file.  When I copy and paste, nothing shows.

aivarannamaa, 2024-02-10:
Try removing Thonny, then search your home directory for `backend.py`:

` find ~/ -name "backend.py"`

Does this command return anything?

eugenerod33, 2024-02-12:
I was unable to execute the command.  When I tried, I got the "permission denied" message.  When I tried to use "sudo" before the command, I got a list of usage messages, with which I am not familiar.  
By the way, I have seen similar messages to the one I got in Thonny while using IDLE, so the fault may lie in the python code.
Eugene Rodriguez

eugenerod33, 2024-02-13:
I retried it, removing the space between '/' and '~name' and the result was:
No such file or directoryEugene Rodriguez
-- 
 Sent with Tuta; enjoy secure & ad-free emails: 
 https://tuta.com

aivarannamaa, 2024-02-13:
Did you copy and paste this command to your Terminal?

eugenerod33, 2024-02-13:
Yes!

aivarannamaa, 2024-02-13:
Alright. After you uninstalled Thonny, what happens when you start your system's `python3` and execute `import thonny`?

eugenerod33, 2024-02-13:
ModuleNotFoundError: No module named 'thonny'

aivarannamaa, 2024-02-13:
When you now reinstall Thonny 4.1.4, does the problem persist?

eugenerod33, 2024-02-26:
I've not seen the message again.
Thanks
Eugene Rodriguez

aivarannamaa, 2024-02-27:
Great!

closed 2024-02-27 by aivarannamaa