thonny/thonny #2993: pylint KeyError: 'R0022' when running any script

SteveDaulton, 2023-11-05:
Thonny 3.3.14 on Xubuntu 22.04.3 LTS.
After running any Python script in Thonny, this error message is displayed:

```
Traceback (most recent call last):
  File "/usr/lib/python3.10/tkinter/__init__.py", line 1921, in __call__
    return self.func(*args)
  File "/usr/lib/python3.10/tkinter/__init__.py", line 839, in callit
    func(*args)
  File "/usr/lib/python3/dist-packages/thonny/ui_utils.py", line 2139, in poll
    on_completion(proc, out_lines, err_lines)
  File "/usr/lib/python3/dist-packages/thonny/plugins/pylint/__init__.py", line 92, in _parse_and_output_warnings
    check = checks_by_id[atts["msg_id"]]
KeyError: 'R0022'

```

Example  script:
```
print("hello world")

```

SteveDaulton, 2023-11-05:
Workaround:
Add the following line to "Tools > Options > Assistant"
```
bad-option-value
```

aivarannamaa, 2023-11-19:
Should be fixed in https://github.com/thonny/thonny/releases/tag/v4.1.4

closed 2023-11-19 by aivarannamaa