thonny/thonny #2995: Micropython download logic is broken

George3d6, 2023-11-12:
When I try to download it I get the following prompt:

```
Downloading 609792 bytes from https://micropython.org/resources/firmware/rp2-pico-20220618-v1.19.1.uf2

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/uf2dialog.py", line 271, in _perform_work
    self._download_to_the_device(download_url, size, target_dir)
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/uf2dialog.py", line 333, in _download_to_the_device
    with urlopen(req, timeout=5) as fsrc:
  File "/usr/lib/python3.10/urllib/request.py", line 216, in urlopen
    return opener.open(url, data, timeout)
  File "/usr/lib/python3.10/urllib/request.py", line 525, in open
    response = meth(req, response)
  File "/usr/lib/python3.10/urllib/request.py", line 634, in http_response
    response = self.parent.error(
  File "/usr/lib/python3.10/urllib/request.py", line 563, in error
    return self._call_chain(*args)
  File "/usr/lib/python3.10/urllib/request.py", line 496, in _call_chain
    result = func(*args)
  File "/usr/lib/python3.10/urllib/request.py", line 643, in http_error_default
    raise HTTPError(req.full_url, code, msg, hdrs, fp)
urllib.error.HTTPError: HTTP Error 404: Not Found
``` worse yet, there is not fallback recommended.

George3d6, 2023-11-12:
Update: this is due to Ubuntu having an oudated version of thonny.

As far as I can tell though the core issue persists in this current version in the form of:
1. Outdated hardcoded urls (thonny or micropy should maintain some mirror of sorts so that broken urls aren't used)
2. Deprecation warning that lists the likely root cause of this as being a missing url

Even worse, the error message doesn't properly display in i3 for some reason (this might be more of an i3 issue) -- the install happens in the background, afaict, but no error is ever surfaced, leading a very confusing UX.

Would you be willing to accept some patches making this software useable for unlucky first users in the future?

pwarias, 2023-11-16:
In case anyone else stumbles into this issue you can work around it by downloading the firmware file manually and uploading it to the Pico. More details can be found here: https://projects.raspberrypi.org/en/projects/get-started-pico-w/1

omani, 2023-11-17:
> Update: this is due to Ubuntu having an oudated version of thonny.

this is not true. it is because the link in https://github.com/thonny/thonny/blob/fb05519c013c3d21e67c8c701b2030b24d47ccc8/data/rpi-pico-firmware.json#L4 is pointing to a resource that is not available (anymore) over at micropython.org. you can check it for yourself:

https://micropython.org/resources/firmware/rp2-pico-20220618-v1.19.1.uf2

omani, 2023-11-17:
someone, anyone. just update the json file and fix this issue.

closed 2023-11-19 by aivarannamaa

aivarannamaa, 2023-11-19:
Thanks for the input!

I now accepted @omani 's PR, which updates the file used by older Thonny versions.

> Would you be willing to accept some patches making this software useable for unlucky first users in the future?

@George3d6, sure! It would be great if you could make this kind of errors more user-friendly.

PR #3005 (2023-11-19): Update rpi firwamre link in json manifest
fixes #2995
commit c0cda30c (2023-11-19): Fixes #2995: Update rpi firwamre link in json manifest (#3005)