thonny/thonny #2996: Thonny crashes when using search functionality

oliviereinaldo, 2023-11-12:
Hello Thonny team,

I'm experiencing an issue with the search functionality in Thonny on Linux Mint 21.1, where the application crashes unexpectedly. Whenever I attempt to use the search feature, Thonny closes abruptly.

Could you please provide guidance on troubleshooting this issue or suggest any specific steps to resolve it? Additionally, I'm using Thonny version 4.1.3. Are there known conflicts or configurations that might lead to this behavior?

Thank you for your assistance.

aivarannamaa, 2023-11-18:
Thank you for the report!

I tried to reproduce this in latest Ubuntu, but could not. Which Tk version do you have? You can check it with Thonny's "Help => About".

dcmcshan, 2024-01-09:
This also happens on Mac OSX 14.0, Python 3.10.11, Tk 8.6.13

Thank you!

oliviereinaldo, 2024-02-24:
Here are the specifications:

Linux:

No LSB modules are available.
Distributor ID: Linuxmint
Description: Linux Mint 21.1
Release: 21.1
Codename: vera

Thonny:

Linux Mint 21.1
Python 3.10.12
Tk 8.6.12

Usually, the error occurs only in code with many lines.

https://github.com/thonny/thonny/assets/129457641/df53d430-a9c0-4302-8475-98bb1cca9778

oliviereinaldo, 2024-02-24:
Here are the specifications:

Linux:

No LSB modules are available.
Distributor ID: Linuxmint
Description: Linux Mint 21.1
Release: 21.1
Codename: vera

Thonny:

Linux Mint 21.1
Python 3.10.12
Tk 8.6.12

Usually, the error occurs only in code with many lines.

https://github.com/thonny/thonny/assets/129457641/c620120a-2933-4120-ac33-5d73d1c07edc

aivarannamaa, 2024-02-25:
Thank you for the screencast!

Can you always repeat the crash with this file? Could you send it to me?

oliviereinaldo, 2024-02-25:
This error occurs with any file that contains many lines of code.

wohltat, 2024-03-09:
I also get crashes when using the search

```
13:16:25.659 [MainThread] ERROR   thonny.plugins.find_replace: Could not get seed text
Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/thonny/plugins/find_replace.py", line 76, in __init__
    selected_lines = widget.selection_get().splitlines()
                     ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/tkinter/__init__.py", line 1029, in selection_get
    return self.tk.call(('selection', 'get') + self._options(kw))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
_tkinter.TclError: PRIMARY selection doesn't exist or form "STRING" not defined
13:16:25.713 [MainThread] INFO    thonny.ui_utils: Parent y: 803, rooty: 803, vrooty: 0
13:16:25.714 [MainThread] INFO    thonny.ui_utils: Placing .!codeview2.!finddialog with geometry +499+1088
[1]    53593 segmentation fault (core dumped)  thonny
```

It seems to happen when there are certain Unicode characters in the text like the following
```
Network symbol: 🖧 Wifi symbol: 🛜,
```

dcmcshan, 2024-05-16:
Here is crash log on OSX from a simple text search:
`-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               Python [35449]
Path:                  /Applications/Thonny.app/Contents/MacOS/Python
Identifier:            org.python.python
Version:               3.10.11 (3.10.11)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-05-16 14:37:16.7973 -0600
OS Version:            macOS 14.4.1 (23E224)
Report Version:        12
Anonymous UUID:        2B45F207-2198-7041-4F3A-DE7B6A81199A

Sleep/Wake UUID:       FF6B54ED-4C18-463D-92AF-96423A436A4E

Time Awake Since Boot: 3000000 seconds
Time Since Wake:       105975 seconds

System Integrity Protection: disabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000136018000
Exception Codes:       0x0000000000000001, 0x0000000136018000

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   Python [35449]

VM Region Info: 0x136018000 is not in any region.  Bytes after previous region: 1  Bytes before following region: 8290304
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      CoreAnimation               133800000-136018000    [ 40.1M] rw-/rwx SM=PRV  
--->  GAP OF 0x7e8000 BYTES
      MALLOC_SMALL                136800000-137000000    [ 8192K] rw-/rwx SM=PRV  

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x191816a60 __pthread_kill + 8
1   libsystem_pthread.dylib       	       0x19184ec20 pthread_kill + 288
2   libsystem_c.dylib             	       0x1917244e8 raise + 32
3   Python                        	       0x103ab0d10 faulthandler_fatal_error + 392
4   libsystem_platform.dylib      	       0x19187f584 _sigtramp + 56
5   libtcl8.6.dylib               	       0x1036eb93c Tcl_UtfToLower + 136
6   libtk8.6.dylib                	       0x103369808 TextSearchAddNextLine + 384
7   libtk8.6.dylib                	       0x103367804 TextSearchCmd + 1628
8   libtk8.6.dylib                	       0x103363c9c TextWidgetObjCmd + 1392
9   libtcl8.6.dylib               	       0x1036104b8 Tcl_EvalObjv + 340
10  _tkinter.cpython-310-darwin.so	       0x10310cdd8 Tkapp_Call + 336
11  Python                        	       0x10387843c method_vectorcall_VARARGS + 124
12  Python                        	       0x10399f140 call_function + 132
13  Python                        	       0x10399647c _PyEval_EvalFrameDefault + 19688
14  Python                        	       0x10398ff28 _PyEval_Vector + 360
15  Python                        	       0x10386cc64 method_vectorcall + 288
16  _tkinter.cpython-310-darwin.so	       0x10310edac PythonCmd + 196
17  libtcl8.6.dylib               	       0x1036104b8 Tcl_EvalObjv + 340
18  _tkinter.cpython-310-darwin.so	       0x10310cdd8 Tkapp_Call + 336
19  Python                        	       0x10387843c method_vectorcall_VARARGS + 124
20  Python                        	       0x10399f140 call_function + 132
21  Python                        	       0x10399647c _PyEval_EvalFrameDefault + 19688
22  Python                        	       0x10398ff28 _PyEval_Vector + 360
23  Python                        	       0x10386cbc0 method_vectorcall + 124
24  Python                        	       0x10399f140 call_function + 132
25  Python                        	       0x103995be0 _PyEval_EvalFrameDefault + 17484
26  Python                        	       0x10398ff28 _PyEval_Vector + 360
27  Python                        	       0x10399f140 call_function + 132
28  Python                        	       0x10399647c _PyEval_EvalFrameDefault + 19688
[... 845 lines cut ...]
      "deploymentId" : 240000031
    }
  ],
  "experiments" : [

  ]
}
}

Model: MacBookPro18,4, BootROM 10151.101.3, proc 10:8:2 processors, 32 GB, SMC 
Graphics: Apple M1 Max, Apple M1 Max, Built-In
Display: Color LCD, 3024 x 1964 Retina, Main, MirrorOff, Online
Memory Module: LPDDR5, Hynix
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4387), wl0: Jan 20 2024 04:08:41 version 20.103.12.0.8.7.171 FWID 01-e09d2675
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: CP2102N USB to UART Bridge Controller
USB Device: USB31Bus
USB Device: USB31Bus
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
`

Unrud, 2024-08-22:
The crash occurs when the document contains any Unicode symbol that is >0xFFFF.

```python
with open("crash.txt", "w", encoding="utf-8") as f:
    f.write(chr(0x10000))
```

Unrud, 2024-11-26:
It's a bug in Tcl 8. The problem can be avoided by compiling Tcl/Tk with `-DTCL_UTF_MAX=6`.

```c
#include <tcl.h>
int main() {
  char str[] = {0xf0, 0x90, 0x80, 0x80, '\0'};
  Tcl_UtfToLower(&str[0]); // Buffer overflow
  return 0;
}
```

I guess this issue can be closed as it's not a bug in thonny.