thonny/thonny #3001: ManagementError when reset

puppet13th, 2023-11-15:
hi there, i have using thonny version 4.1.3 and been playing with deepsleep and lightsleep on esp32. sometimes when i do hardreset( using en reset button) or deepsleep thonny just this error:

```
MPY: soft reboot
reset cause : 5
wake reason : 2
MicroPython v1.21.0 on 2023-10-25; LOLIN D32 PRO NO SPIRAM with ESP32

Type "help()" for more information.

>>> PROBLEM IN THONNY'S BACK-END: Exception while handling 'get_dirs_children_info' (thonny.plugins.micropython.mp_back.ManagementError: Management markers missing).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 1.
```

is there any config in thonny to prevent that? maybe something like delay or timeout?
[backend.log](https://github.com/thonny/thonny/files/13360266/backend.log)

aivarannamaa, 2023-11-18:
Thank you for the report!

This is similar to #2748 and #2833.

Can you give me the code that causes this?

puppet13th, 2023-11-19:
```py
import asyncio

def counter( num ):
    count = num
    while True:
        count += num
        print( count )
        await asyncio.sleep( 1 )

event_loop = asyncio.new_event_loop()
task_list = []
for num in range( 1,9 ):
    task_list.append( event_loop.create_task( counter( num ) ) )

event_loop.run_forever()
```

when the code is running, pressing reset will do the trick.

```
28
32
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0xed8
load:0x403cc710,len:0x714
load:0x403ce710,len:0x2c24
entry 0x403cc710
wake  : 0
reset : 1
KeyboardInterrupt?
MicroPython v1.21.0 on 2023-10-25; ESP32C3 module with ESP32C3
Type "help()" for more information.
>>> PROBLEM IN THONNY'S BACK-END: Exception while handling 'get_globals' (thonny.plugins.micropython.mp_back.ManagementError: Script produced errors).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 1.
```

i add some delay in boot.py. same issue if i remove delay from boot.py
```py
# This file is executed on every boot (including wake-boot from deepsleep)
#import esp
#esp.osdebug(None)
#import webrepl
#webrepl.start()

from machine import Pin
Pin( 19, Pin.OUT, value=0 )

from machine import wake_reason, reset_cause
from time import sleep
from os import listdir

def wake():
    print( 'wake  :', wake_reason() )
    print( 'reset :', reset_cause() )
wake()
    
try:
    print( 'KeyboardInterrupt?' )
    sleep( 3 )
    if 'main.py' not in listdir( '/' ):
        print( 'main.py' )
    else:
        try:
            import demo
        except:
            pass
        
except KeyboardInterrupt:
    print( 'KeyboardInterrupt!' )
```

[backend.log](https://github.com/thonny/thonny/files/13402082/backend.log)