thonny/thonny #3010: Dedicate next version of Thonny to victims of war crimes in Israel and Gaza

jrmoserbaltimore, 2023-11-19:
Thonny 4.0 was dedicated to Ukraine in defending itself from the invasion by Russia.

Can we dedicate the next version of Thonny to the victims of war crimes in Israel (by Hamas) and Gaza (by Netanyahu's administration)?  The greatest majority of their citizens continue to protest the actions of their respective governments; it is critical we remember that the actions of a country are not the actions of its people, especially in these times when violence against people solely because of their ethnicity is on the rise.

oasis4cns, 2023-11-20:
If we say Hamas we have to say on the other hand Israel army, or Hania administration and Netanyahu.
Best regards

jrmoserbaltimore, 2023-11-20:
Fine by me, dump it squarely on leadership here and clearly _not_ on the people of Israel and Palestine.  This is not the fault of innocent civilians and that's worth pointing out directly.

aivarannamaa, 2023-11-20:
> Can we dedicate the next version of Thonny to the victims of war crimes in Israel (by Hamas) and Gaza (by Netanyahu's administration)? 

After Russian troops leave Ukraine I'll consider helping other causes, but until then I'll focus on Ukraine.

closed 2023-11-20 by aivarannamaa

webdeb, 2024-02-23:
There are a lot things going on in the world, and it is great, that @aivarannamaa offers a political platform in software. Also there is no reason to support only one topic, there is plenty of space in the bar next to the Ukraine icon, and we should use it. https://github.com/thonny/thonny/issues/3122 (I like Pandas)

I already see all those topics pop up: Ukraine, Gender identity, Palestine Israel War, China Taiwan, Tsunami of the Baltic sea, Climate change, you name it!

I only hope that the author is not a fascists who declares his own view over other users, by using his ultimate github repository superpower, because we should decide democratically by voting.