thonny/thonny #3011: Varibales created offline

oasis4cns, 2023-11-20:
I am using Thonny for teaching and I have seen that when executing scripts (not console commands) variables are not displayed only at the end of execution of all the script code. So, it is suitable to show each variable in realtime.
Best regards

aivarannamaa, 2023-11-20:
Thonny runs programs in single thread, which means that it can't query the variables while the program is running. 

If you want to get updates about changing variables, you need to debug your program instead of running it.

aivarannamaa, 2023-11-20:
I could update the variables when the program is waiting for input, but it would be an arbitrary exception, so I don't like this idea.