thonny/thonny #3012: clear varibales command

oasis4cns, 2023-11-20:
I think Thonny has possibility to clear variables. So it is very useful to add a menu command and a keyboard shortcut for that
best regards

aivarannamaa, 2023-11-24:
If you could tell me more about this wish, I could offer a solution. I which situation do you want to clear variables?

oasis4cns, 2023-11-24:
hi,
for example while teaching, I use variables sidebar to show to students how
variables are changed. But when I start a new shell example, I clear shell
(ctrl+L) but I couldn't clear variables so we don't like to show variables
from the old example with variables of the new example. It is like when
working on Matlab environment.

aivarannamaa, 2023-11-24:
In this case you could press "Stop / Restart" button (or the same command from the Tools menu).

oasis4cns, 2023-11-24:
thanx! stop/restart combine clear shell and clear variables. but it is not
easy to discover that

aivarannamaa, 2023-11-26:
> it is not easy to discover that

As I see it, clearing variables is just a side effect of restarting the interpreter. I don't see why someone would want the former without the latter. Therefore I'd leave current command name as it is.

closed 2023-11-26 by aivarannamaa