thonny/thonny #3014: Management of virtual environments is not documented properly

monga, 2023-11-22:
I'm struggling to understand how Thonny is supposed to deal with virtual environments and I needed to grep the source code to understand better. In particular:

- No menu entry seems to be related to virtual environment
- The options related to the Interpreter can be invoked in three different ways (Tools/Options/Interpreter, Run/Configure interpreter..., Main window bottom right click)
- The dialog mentions that (navagation -> navigation)

```
        venv_text = tr(
            "You can activate an existing virtual environment also"
            + " via the right-click context menu in the file navagation"
            + " when selecting a virtual environment folder,"
            + " or the 'pyveng.cfg' file inside."
        )
```

but I didn't find how to activate the "right-click context menu" in the file navigation (is this the dialog for File/Open... ?): it does not work on my Linux machine nor on a Win10 box where I tried (both with a pip installed Thonny, 4.1.3).

I guess activation of a virtual environment in Thonny boils down to change the interpreter, but this is definitely not documented enough, and I found definitely not novice friendly the fact that when you start Thonny after having worked with a virtual environment interpreter, the interpreter is not the system one by default.

I believe virtual environments are an important issue even for novices, especially if they are learning scientific computing where the use of third-party libraries is very common (numpy, etc.). Learning to install everything system-wide is a bad habit that causes lots of nasty problems hard to diagnose for novices.

aivarannamaa, 2023-11-24:
Thank you for the report!

> No menu entry seems to be related to virtual environment

That's right. I'm considering adding one into the Tools-menu in Thonny 5.0

> "You can activate an existing virtual environment..."

This appears at "Tools => Options => Interpreter => Local Python 3" and refers to the file browser ("View => Files").

[image]

I'll fix the wording in Thonny 5.0

> I found definitely not novice friendly the fact that when you start Thonny after having worked with a virtual environment interpreter, the interpreter is not the system one by default.

Sorry, I don't understand this. Do you mean that when you have activated a virtual environment in Terminal and start Thonny, you would like to have this environment's interpreter to be selected in Thonny as well?

monga, 2023-11-25:
Now I understand, thank you! The missing point for me was the "View => Files" menu item... So probably it is just a problem of the wording of the message.

monga, 2023-11-25:
> Sorry, I don't understand this. Do you mean that when you have activated a virtual environment in Terminal and start Thonny, you would like to have this environment's interpreter to be selected in Thonny as well?

This is a minor point, but a bit confusing for beginners: 

- Thonny is installed system-wide
- Launch Thonny with the system interpreter
- Activate a virtual environment to work on a specific project (this switch the interpreter)
- The next time you launch Thonny is still in the virtual environment, so you have to be careful to switch back the interpreter to the system one if you don't want it

I would prefer Thonny starting always with the system interpreter.

sgllama, 2023-12-08:
> I would prefer Thonny starting always with the system interpreter.

As a counter, I am *very* glad that Thonny has started up in the same state that it was when it was closed. I find that to be a more consistent and less confusing result.

It could easily and quickly become irritating to have to remember *why* the Python code, that Thonny also re-loaded and displayed to me when restarted, is no longer functioning but instead throwing up errors that I'm *sure* had all been dealt with. Then having to navigate the "Files" view away from the project's source code to where the venv lives, activate it and try again. Especially as I expect to be using the same venv on the PC for at least a couple of months, switching only between that venv and CircuitPython for a few microcontroller projects.

ALTHOUGH I would suggest that placing the information about the current interpreter at the bottom RHS of the GUI is a bit out of the way: when Thonny starts I do not find that bottom RHS is where your eye immediately goes, and you actually have to carefully read the path to python.exe in order to spot that, yes, you are in a venv (and which one). Meanwhile (at the moment, as a newbie to Thonny) there is all that space to the RHS of the toolbar, which is just above the name of the project source file I am selecting to work on next...

Aside: I write this as someone who, although already a coder in C/C++, has only been attempting to use Python and Thonny for the past week or so - most of which has been spent reading up on strange new stuff, like tutorials leading me to create a venv (huh? what is a "Python Virtual Environment"? Read, read, read...).

rootsmusic, 2023-12-09:
@sgllama You may be interested in issue #2466

aivarannamaa, 2023-12-26:
Sorry for the late reply! Just like @sgllama, I believe that remembering the last interpreter is the most natural choice for most users, so I don't want to change this.

Regarding the visibility of the interpreter information -- the lower right corner of the interpreter switcher was inspired by PyCharm and I think there are more IDE-s which put it there. The interepreter is also described by the start-up banner in the Shell, so I'm not convinced anything needs to be changed here.

FMaz008, 2024-04-10:
I was also very confused by the instructions. Adding a mentionned to (View ->Files) somewhere would greatly help

jimeggleston, 2024-05-07:
I agree with @FMaz008

jose1711, 2024-07-11:
also `pyveng.cfg` seems to be a typo

IanD42, 2024-08-14:
Thanks for the clarification, I was having the same issues in enabling an existing virtual environment within Thonny (pi zer0) until reading the above
Thanks

aivarannamaa, 2024-10-12:
All this will be updated in Thonny 5.0

closed 2024-10-12 by aivarannamaa