thonny/thonny #3015: Expo Virus in Thonny Lib ?

GuidoAmschler, 2023-11-22:
I recently installed Thonny from thonny.org.  version Windows 11, Python 3.10.11 Tk 8.6.13  

Now ClamWin Antivirus full scan  claims there is an Expiro Virus version hidden in \Thonny\Lib\venv\scripts\nt\pythonw.exe 

Can somebody crosscheck 
 
[image]

aivarannamaa, 2023-11-22:
I now uploaded this file from Thonny 4.1.4 installation to VirusTotal and it looks like ClamWin is the only virus analyzer complaining about this file: https://www.virustotal.com/gui/file/1c37c6c3c14074b69b3b8543ed5edcf6666b876c8f73672e52b58f6bbc5dd3ee

It also shows the file is signed by Python Software Foundation, i.e. not tampered by someone else.

I found a forum post, which also complains about this virus in pythonw.exe https://www.bleepingcomputer.com/forums/t/787136/msiexecexe-in-the-syswow64-detected-as-malware/

Most likely ClamWin is wrong this time.

aivarannamaa, 2023-11-22:
You could try installing official Python 3.10.11 for 64-bit Windows, which has the same file at the same location. If ClamWin reports this file as infected (most likely it will, unless it somehow considers context as well), you could raise an issue at https://github.com/python/cpython/issues. Maybe then we'll get an explanation.

GuidoAmschler, 2023-11-22:
Hi

ClamWin scans take  a very very ! long time  ( I assume due to very deep
history virus signature files )  so  the HD scan that started around noon Geneva time
on my Thinkpad X1  and which claimed to have found  'Expiro"  did not even   terminate now at 20:49h  

If I find time time I  may try your suggestion tomorrow and hopefully  ClamWin  triggered accidentally.

-Hans