thonny/thonny #3017: add Github Discussions

rootsmusic, 2023-11-23:
Several of this repo's "issues" don't seem to be bug reports or feature requests.  If this repo adds Discussions, then users can ask questions and others can answer too.  [Github Discussions](https://docs.github.com/en/discussions/collaborating-with-your-community-using-discussions/about-discussions) offers many features, including moderation.

aivarannamaa, 2023-11-24:
Thanks for the suggestion!

Done now: https://github.com/thonny/thonny/discussions

closed 2023-11-24 by aivarannamaa

rootsmusic, 2023-11-26:
You're welcome!  I suggest linking it on your homepage, so users can find it instead of opening issues for questions.  You may also consider using your [Google Groups](https://groups.google.com/forum/#!forum/thonny) to collaborate with developers.