thonny/thonny #3018: Micropython libraries

thupkens, 2023-11-23:
Cannot install **micropython** libraries from tools -> manage packages menu anymore. Worked well in version 4.0.2.
I use Windows 10 and Thonny 4.1.4.

aivarannamaa, 2023-11-23:
> Cannot install micropython libraries from tools -> manage packages menu anymore. 

What does it mean precisely? Do you get an error message?

pumpkinking10, 2023-11-27:
I have the same problem after installing numpy. Even after installing it using manage packages menu the program still shows error saying numpy is not recognised. I know that it was installed as it was shown on the <INSTALL> menu.
[image]
[image]
The only thing that was off was that I had this message before installing numpy.
[image]

pumpkinking10, 2023-11-27:
If it helps I am using thonny to control motor through a servo2040 controller. And I selected MicroPython(Raspberry Pi Pico) as the interpreter. This is selected when I used manage package to download numpy.

aivarannamaa, 2023-11-27:
I'm surprised that you were able to install numpy on a MicroPython device, as it is quite large package and not meant for MicroPython.

pumpkinking10, 2023-11-28:
Does this mean I should find a work around code and not use numpy? And does this mean it actually didn't installed numpy package? servo2040 controller have only 2MB so I guess that wouldn't be enough.

aivarannamaa, 2023-11-28:
I'll check later what exactly happens when I try to install numpy on a device wit unsufficient flash.

There is a numpy alternative for small devices called ulab, but i don't know if it must be built into MicroPython or can be installed separately.

pumpkinking10, 2023-11-28:
Yeah I am wondering if it can be installed separately as well because I am also using scipy.integrate.solve_ivp to integrate a function. I run the code in my computer instead of on servo2040 and it runs fine but when I switch the interpreter to Pi Pico it doesn't recognise numpy and scipy. I can get around numpy with ulab but I have no idea about scipy.integrate.

pumpkinking10, 2023-12-01:
Actually, The servo2040 controller that I am using have 127MB storage so clearly it can house numpy. So I don't know why I can't use it after downloading it with Manage Package.

Fabri54, 2023-12-04:
I've a problem when trying to instal a package from a location in my Mac.  If from "Manage Packages" I chose "Install from local file" and I click   "here"  Thonny crashes immediately and I can send a nice report to Apple.
Please note that  "Install from PyPi" works.
My Mac runs  Sonoma 14.1.2  on an M2 Pro chip

closed 2025-05-27 by aivarannamaa