thonny/thonny #3029: TC port aren't liberated

moonsac, 2023-12-04:
Hello,
I launched a simple web page with Flask. The first time it works fine. After I stop the script, the TCP port isn't liberated, so new runs are possible but the browser can't load the page on the address localhot:5000 anymore (loading never stops). I had to kill manually all the processes generated by the first launch to free the 5000 port. I tried other IDEs (Edupython for example) and they don't have this issue

aivarannamaa, 2023-12-04:
Thank you for the report!

> After I stop the script

How did you do this?

moonsac, 2023-12-04:
The red Stop button in Thonny

aivarannamaa, 2023-12-04:
The red Stop button kills the process started with "Run current script". If Flask has started new processes, which are not tied to the parent process, then there's nothing Thonny can do in this situation.

What happens, when you close the app with Ctrl-C?

Can you give me the minimal example code and exact steps for reproducing this problem?

Which OS and Thonny version are you using?

moonsac, 2023-12-04:
KeyboardInterrupt doesn't solve the problem. I can't close with Ctrl-C. The code is below.

```
from flask import Flask, render_template

app = Flask(__name__)

@app.route('/')
def index():
  return render_template('index.html')

app.run(debug=True)
```
index.html is a simple hello world page

Thonny 4.0.1
Windows 11 (64-bit)
Python 3.10.6 (64-bit)