thonny/thonny #3030: Feature request: Reload a file

OlegAlexander, 2023-12-05:
Hello and thank you for creating Thonny!!

I'm using Thonny to teach my son Python. A common programming task is to save some data to a json or csv file from within Thonny (internal file change, not external). Currently, the only way to reload a json/csv file is to close it and open it again.

Is it possible to either:

- Automatically reload a file if it's been changed when the user clicks on that file's tab?
- Or, add a Reload option under the right-click menu for a file's tab next to Close, Close Others, and Close All, as well as the File menu?

Thank you!

aivarannamaa, 2024-03-17:
Which Thonny version are you using?

I had forgotten, that Thonny already is supposed to check for file changes after an execution completes.

OlegAlexander, 2024-03-19:
Hello my friend. I just tested this in Thonny 4.1.4 and the issue has been solved! I tested writing a text file from inside Thonny and every time I changed the text, the tab with the text file was updated. Same with csv files. 

I haven't tested writing to a text file outside of Thonny and seeing if its tab changed. But if that doesn't work, it's not a big deal because the main use case is writing to files from inside Thonny. 

I will close this issue. Thank you so much for fixing this and for creating Thonny!

closed 2024-03-19 by OlegAlexander