thonny/thonny #3031: pgzero not found

leighvWall, 2023-12-05:
hi
Am on Thonny 4.1.3 - and trying to use pgzero

I've installed Pygame from the plug-in window (1.2.1) and it seems to have all the dependencies ok.

I've tried running the code below (with Pygame Zero mode ticked):

```
WIDTH = 300
HEIGHT = 300

def draw():
    screen.fill((128, 0, 0))
```

However, the message I get each time is:

_WARNING: Pygame Zero mode is turned on (Run → Pygame Zero mode),
but pgzero module is not found. Running program in regular mode._

Is there something I've missed about how to run Pygame Zero from Thonny?

thanks

Vernon

aivarannamaa, 2023-12-06:
You also need to install `pgzero`, `pygame` is not enough.

tdegouw, 2024-02-28:
For those arriving here with this question: Try installing through the "tools -> manage packages" menu option instead of the "tools -> manage plug-ins" option.