thonny/thonny #3033: Problem with Thonny “save”

JoKesCode, 2023-12-06:
| I am a new user and have a new Raspberrry Pi 5 and runnig it on my Mac with TigerVNC. Now I try to work with Thonny (4.1.4). But it is not possible to save files. When I select „save“ or "save as ..“ there is no possibility to give a name. I only see the folders of my raspberry and no possibility to save the file. I then only can close the folder window by pressing „esc“. What can I do to solve this problem?
-- | --

aivarannamaa, 2023-12-06:
Can you post a screenshot describing your problem?

JoKesCode, 2023-12-06:
> Thank you very much for the quick response. 
> 
> Attached I send some screenshots. It is not possible to save one file or open another. There is no way to enter a file name. Even the immediately opened file manager does not allow saving. Here I can only come back with the "ESC" key. Hope you have a solution for me. 
> 
> Best regards, Jochen Kempkens
￼
￼
￼
￼

aivarannamaa, 2023-12-06:
Unfortunately the images didn't come through. It looks GitHub doesn't support images via email. Please post them via web browser.

JoKesCode, 2023-12-07:
[image]

[image][image]
[image]
[image]

aivarannamaa, 2023-12-07:
Thanks! Now I understand.

It looks like the file selection app (zenity) by default creates too large windows for your resolution. You could try resizing it (press Alt, click anywhere on the window and drag until you see a corner of the window).

aivarannamaa, 2023-12-07:
If this doesn't help, then in Thonny go to "Tools => Options => General" and select "Use Tk file dialogs instead of Zenity"

aivarannamaa, 2023-12-07:
> Both options work. The second is more convenient. 😄

According to my tests, Zenity remembers the desired window size after you have resized it once.

closed 2023-12-07 by aivarannamaa