thonny/thonny #3036: Slow mouse cursor when switching between windows (RPI 5)

samneggs, 2023-12-13:
I was using Raspberry PI 4 and Thonny version 3. I upgraded to a PI 5 and Thonny v4. I noticed a obvious and annoying lag or slowdown when moving mouse between menu and windows within Thonny.  The GPU usage spikes to 80-90 %
Version 4.1.4
Debian GNU/Linux 12(bookworm)
Raspberry PI 5
6.1.0-rpi7-rpi-2712

samouche6453, 2023-12-16:
I'm also having the mouse cursor movements lagging in the Tksheet's spreadsheet/tables window area, but not outside of the sheet window.

aivarannamaa, 2023-12-26:
Thank you for the report and sorry for the late reply!

Bookworm on Raspberry 5 switched from X-Server to Wayland. Thonny's UI toolkit is based on Tk, which does not support Wayland directly, therefore RPi uses [Xwayland](https://wayland.freedesktop.org/xserver.html) for running Thonny. I suppose this causes the lag. If Tksheet (which is also based on Tk) also causes this lag, then I'm afraid the solution is out of my reach. It may be worth contacting RPi developers -- maybe they can find a solution.

chedstrom, 2023-12-29:
> Thank you for the report and sorry for the late reply!
>
> Bookworm on Raspberry 5 switched from X-Server to Wayland. Thonny's UI toolkit is based on Tk, which does not support Wayland directly, [...]

Thank you for the explanation! i had also run into this problem recently after updating to Bullseye to a 4 on Bookworm. I thought maybe it was because i was running on a larger monitor which happened at the same time, except everything else was running smoothly.

BrianUsher, 2024-05-26:
Did you ever fix this issue? I'm running a Raspberry Pi 4B. I developed by code on a Raspberry Pi 3B and it worked well on the 3B. I changed over to a 4B and I'm experiencing the same lag etc GPU >90% and the performance running my Python code on the 4B is rubbish compared to the 3B. I have the same results on a Pi 5. Any help would be appreciated.

chedstrom, 2024-05-26:
I did not, unfortunately for Bookworm, and it was only ever thonny. I read in another forum somewhere it was because of how thonny was programmed so that it’s not ideal for Bookworm. My solution was to run Bullseye on my Pi 4, as it wasn’t critical for me to run Bookworm.