thonny/thonny #3038: ESP32: Thonny's backend output in Micropython WebREPL.

IhorNehrutsa, 2023-12-14:
There is Thonny's backend output in Micropython WebREPL when the ESP32 is connected to Thonny and WebREPL at the **same time**.
Please hide it.
Thank you.
[image]

aivarannamaa, 2023-12-26:
There is no way to hide it if you want to use two clients at the same time. MicroPython reuses same REPL for both serial and WebREPL connections. Actually I'm surprised that you can even use two connections at the same time. Most likely you will meet problems at some point.

closed 2023-12-26 by aivarannamaa

IhorNehrutsa, 2023-12-26:
If REPL output is raw, there are no restrictions to use two clients at the same time.
[image]
[image]