thonny/thonny #3039: Autocomplete incomplete module list

worksonmymachine-de, 2023-12-14:
Hi! I want to address an issue I have with Thonny 4.1.4 on Windows10 and Raspberry Pi Pico W with MicroPython 1.21. I noticed it since some time but with the current versions this still happens.

In my case the module aioble is not shown in the autocomplete list triggered by CTRL+SPACE.
I can write the import statement unsupported 'import aioble' and when running the script no ImportError occurs.
Additionally when calling help('modules') the following list is shown.

[image]

When imported the autocomplete for "aioble." doesn't show when pressing CTRL+SPACE - this is probably a follow up error as the module itself is unknown.

Is this a jedi issue? I saw a [similar issue](https://github.com/thonny/thonny/issues/2670) which was postponed to v5.
For many other modules and local variables the autocomplete works totally fine.
Can I trigger a refresh / rebuild of the module list Thonny / jedi uses for autocompletion?

aivarannamaa, 2023-12-26:
Thank you for the report!

Thonny's MicroPython autocompletion is based on a directory of module stubs present in Thonny's program folder. Currently I've included only the stubs of the modules documented at https://docs.micropython.org/en/latest/. Jedi uses this directory for autocompletion proposals.

I have plans to allow the user to submit their custom modules and stubs for autocompletion purposes (#2186) but I haven't got to it yet.

aivarannamaa, 2025-05-27:
In Thonny 5.0 you will get more completions by default and you will be able to install your own set of packages for completion purposes.

closed 2025-05-27 by aivarannamaa