thonny/thonny #3043: Microbit 2.2  Misleading error messages and wrong results when using Thonny instead of WebIDE

rkompass, 2023-12-18:
Hello @aivarannamaa,

Before beginning I have to say a big thank you for your efforts to make Thonny a nice slim IDE.
I use it every day and mostly don't experience problems.

Now I got a problem however, that first looked like a problem of the Micropython firmware:
I compiled from this [github site](https://github.com/microbit-foundation/micropython-microbit-v2) and afterwards could flash the image to eh Microbit 2.2 (moving the firmware to the MICOBIT drive). With this method I didn't experience the connection problems I had before: Thonny couldn't connect with the firmware copied to MAINTENANCE drive (as told in the instructions).

Now everything going smooth I tried e.g.

```py
@micropython.asm_thumb
def asm_add(r0, r1):
    add(r0, r0, r1)

x = asm_add(4, 6)

print(x)
```
and got a `SyntaxError: expecting an assembler instruction` at the `add(r0, r0, r1)` position.

With another example:
```py
def generator_function():
    yield 42
    yield 22

generator = generator_function()

# generator yields only 42
for value in generator:
    print(value)
```
The output is
```
None
None
```
instead of
```
42
22
```
The generator spitting out `None` problem only occurs if I run the example code from file (F5), with copy and paste it prints the correct 42 and 22.

aivarannamaa, 2023-12-26:
Thank you for the report and sorry for the late reply!

> Thonny couldn't connect with the firmware copied to MAINTENANCE drive (as told in the instructions).

This issue was probably about terminology -- depending on the context "firmware" may mean MicroPython or something else. In this case it was something else -- in the micro:bit world, firmware is a piece of program, which operates below MicroPython layer. MicroPython sits at the application layer, with py-files being the configuration of this application (even though Python programmers may see it otherwise).

The main issue is related to #1441 and #1718. I'll try to find a better solution.

aivarannamaa, 2023-12-26:
I was able to fix this. The fix will be released in the first beta of Thonny 5.0.

closed 2023-12-26 by aivarannamaa

rkompass, 2023-12-29:
Big thanks and have a happy new year!