thonny/thonny #3044: PROBLEM IN THONNY'S BACK-END: ...

aaronacj, 2023-12-18:
ESP32-S3-Zero (https://www.amazon.com/dp/B0CHSBQ1G4?psc=1&ref=ppx_yo2ov_dt_b_product_details)
Thonny 4.1.4 (2023-11-19)

Can't save files (main.py or boot.py...)

-------------------------------------------------------
Error:

PROBLEM IN THONNY'S BACK-END: Exception while handling 'write_file' (OSError: Could not open file /main.py for writing, output:
Traceback (most recent call last):

  File "<stdin>", line 4, in <module>

OSError: [Errno 19] ENODEV).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 1.

-------------------------------------------------------
Backend:

14:37:38.201 [MainThread] INFO    thonny: Thonny version: 4.1.4
14:37:38.201 [MainThread] INFO    thonny: cwd: C:\Users\aaron
14:37:38.201 [MainThread] INFO    thonny: original argv: ['C:\\Program Files (x86)\\Thonny\\python.exe', '-s', '-u', '-B', 'C:\\Program Files (x86)\\Thonny\\lib\\site-packages\\thonny\\plugins\\micropython\\bare_metal_backend.py', "{'clean': False, 'port': 'COM10', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': True, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'GenericBareMetalMicroPythonProxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
14:37:38.201 [MainThread] INFO    thonny: sys.executable: C:\Program Files (x86)\Thonny\python.exe
14:37:38.201 [MainThread] INFO    thonny: sys.argv: ['C:\\Program Files (x86)\\Thonny\\lib\\site-packages\\thonny\\plugins\\micropython\\bare_metal_backend.py', "{'clean': False, 'port': 'COM10', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': True, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'GenericBareMetalMicroPythonProxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
14:37:38.201 [MainThread] INFO    thonny: sys.path: ['C:\\Program Files (x86)\\Thonny\\lib\\site-packages\\thonny\\plugins\\micropython', 'C:\\Program Files (x86)\\Thonny\\python310.zip', 'C:\\Program Files (x86)\\Thonny\\DLLs', 'C:\\Program Files (x86)\\Thonny\\lib', 'C:\\Program Files (x86)\\Thonny', 'C:\\Program Files (x86)\\Thonny\\lib\\site-packages']
14:37:38.201 [MainThread] INFO    thonny: sys.flags: sys.flags(debug=0, inspect=0, interactive=0, optimize=0, dont_write_bytecode=1, no_user_site=1, no_site=0, ignore_environment=0, verbose=0, bytes_warning=0, quiet=0, hash_randomization=1, isolated=0, dev_mode=False, utf8_mode=0, warn_default_encoding=0, int_max_str_digits=-1)
14:37:38.201 [MainThread] INFO    thonny.plugins.micropython.bare_metal_backend: Starting backend, args: {'clean': False, 'port': 'COM10', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': True, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'GenericBareMetalMicroPythonProxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}
14:37:38.201 [MainThread] INFO    thonny.plugins.micropython.bare_metal_backend: Initial submit_mode: raw_paste, write_block_size: 127, write_block_delay: 0, read_block_size: 0
14:37:38.201 [MainThread] INFO    thonny.plugins.micropython.mp_back: Initializing MicroPythonBackend of type GenericBareMetalMicroPythonBackend
14:37:38.201 [MainThread] INFO    thonny: TIME/MODS 0.125 before prepare (+68 modules)
14:37:38.201 [MainThread] INFO    thonny: NEW MODS ['__future__', '_bz2', '_compression', '_ctypes', '_datetime', '_heapq', '_lzma', '_queue', '_tkinter', 'bz2', 'ctypes', 'ctypes._endian', 'ctypes.wintypes', 'datetime', 'encodings.ascii', 'errno', 'faulthandler', 'fnmatch', 'gettext', 'heapq', 'lzma', 'math', 'msvcrt', 'pathlib', 'platform', 'posixpath', 'queue', 'serial', 'serial.serialutil', 'serial.serialwin32', 'serial.win32', 'shlex', 'shutil', 'signal', 'subprocess', 'thonny.backend', 'thonny.base_file_browser', 'thonny.codeview', 'thonny.config_ui', 'thonny.editors', 'thonny.languages', 'thonny.misc_utils', 'thonny.plugins', 'thonny.plugins.backend_config_page', 'thonny.plugins.micropython', 'thonny.plugins.micropython.connection', 'thonny.plugins.micropython.mp_back', 'thonny.plugins.micropython.mp_common', 'thonny.plugins.micropython.mp_front', 'thonny.plugins.micropython.serial_connection', 'thonny.plugins.micropython.webrepl_connection', 'thonny.roughparse', 'thonny.running', 'thonny.tktextext', 'thonny.ui_utils', 'thonny.workdlg', 'tkinter', 'tkinter.commondialog', 'tkinter.constants', 'tkinter.dialog', 'tkinter.filedialog', 'tkinter.font', 'tkinter.messagebox', 'tkinter.simpledialog', 'tkinter.ttk', 'urllib', 'urllib.parse', 'zlib']
14:37:38.201 [MainThread] INFO    thonny.plugins.micropython.mp_back: Sending interrupt
14:37:38.201 [MainThread] INFO    thonny.plugins.micropython.mp_back: Done sending interrupt
14:37:38.266 [MainThread] INFO    thonny.plugins.micropython.bare_metal_backend: Poking
14:37:38.313 [MainThread] INFO    thonny: TIME/MODS 0.111 got welcome 
14:37:38.313 [MainThread] INFO    thonny: TIME/MODS 0.000 bef preparing helpers 
14:37:38.313 [MainThread] INFO    thonny.plugins.micropython.mp_back: Preparing helpers
14:37:38.313 [MainThread] INFO    thonny: TIME/MODS 0.000 befsubcode 
14:37:38.313 [MainThread] INFO    thonny.plugins.micropython.bare_metal_backend: Requesting raw mode at b'>>> '
14:37:38.313 [MainThread] INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: '\r\n'
14:37:38.344 [MainThread] INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: ''
14:37:38.344 [MainThread] INFO    thonny.plugins.micropython.bare_metal_backend: Entered raw prompt
14:37:38.360 [MainThread] INFO    thonny: TIME/MODS 0.047 affsubcode 
14:37:38.391 [MainThread] INFO    thonny: TIME/MODS 0.031 affforw 
14:37:38.391 [MainThread] INFO    thonny: TIME/MODS 0.000 prepared helpers 
14:37:38.391 [MainThread] INFO    thonny: TIME/MODS 0.000 befsubcode 
14:37:38.406 [MainThread] INFO    thonny: TIME/MODS 0.016 affsubcode 
14:37:38.422 [MainThread] INFO    thonny: TIME/MODS 0.016 affforw 
14:37:38.422 [MainThread] INFO    thonny: TIME/MODS 0.000 got cwd 
14:37:38.422 [MainThread] INFO    thonny: TIME/MODS 0.000 befsubcode 
14:37:38.438 [MainThread] INFO    thonny: TIME/MODS 0.016 affsubcode 
14:37:38.453 [MainThread] INFO    thonny: TIME/MODS 0.016 affforw 
14:37:38.453 [MainThread] INFO    thonny: TIME/MODS 0.000 prepared 
14:37:38.453 [MainThread] INFO    thonny.plugins.micropython.mp_back: Prepared
14:37:38.453 [MainThread] INFO    thonny: TIME/MODS 0.000 befsubcode 
14:37:38.469 [MainThread] INFO    thonny: TIME/MODS 0.016 affsubcode 
14:37:38.500 [MainThread] INFO    thonny: TIME/MODS 0.031 affforw 
14:37:38.500 [MainThread] INFO    thonny: TIME/MODS 0.000 befsubcode 
14:37:38.516 [MainThread] INFO    thonny: TIME/MODS 0.016 affsubcode 
14:37:38.547 [MainThread] INFO    thonny: TIME/MODS 0.031 affforw 
14:37:38.547 [MainThread] INFO    thonny: TIME/MODS 0.000 befsubcode 
14:37:38.563 [MainThread] INFO    thonny: TIME/MODS 0.016 affsubcode 
14:37:38.578 [MainThread] INFO    thonny: TIME/MODS 0.016 affforw 
14:37:38.578 [MainThread] INFO    thonny.backend: Loading Jedi
14:37:38.578 [MainThread] INFO    thonny: TIME/MODS 0.000 Before loading Jedi 
14:37:38.578 [MainThread] INFO    thonny.common: Using THONNY_FRONTEND_SYS_PATH ['C:\\Program Files (x86)\\Thonny\\python310.zip', 'C:\\Program Files (x86)\\Thonny\\DLLs', 'C:\\Program Files (x86)\\Thonny\\lib', 'C:\\Program Files (x86)\\Thonny', 'C:\\Program Files (x86)\\Thonny\\lib\\site-packages', 'C:\\Users\\aaron\\AppData\\Roaming\\Thonny\\plugins\\Python310\\site-packages']
14:37:38.751 [MainThread] INFO    thonny: TIME/MODS 0.172 After loading Jedi (+123 modules)
[... 16 lines cut ...]
14:37:40.861 [MainThread] INFO    thonny: TIME/MODS 0.019 affforw 
14:37:40.861 [MainThread] INFO    thonny: TIME/MODS 0.000 after get_dirs_children_info 
14:37:45.947 [MainThread] INFO    thonny: TIME/MODS 5.086 before write_file 
14:37:45.947 [MainThread] INFO    thonny: TIME/MODS 0.000 befsubcode 
14:37:45.962 [MainThread] INFO    thonny: TIME/MODS 0.015 affsubcode 
14:37:45.990 [MainThread] INFO    thonny: TIME/MODS 0.028 affforw 
14:37:45.990 [MainThread] INFO    thonny: TIME/MODS 0.000 befsubcode 
14:37:46.006 [MainThread] INFO    thonny: TIME/MODS 0.016 affsubcode 
14:37:46.037 [MainThread] INFO    thonny: TIME/MODS 0.031 affforw 
14:37:46.037 [MainThread] ERROR   thonny.backend: Exception while handling 'write_file'
Traceback (most recent call last):
[... cut ...]
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 1294, in _write_file
    self._write_file_via_serial(source_fp, target_path, file_size, callback)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 1354, in _write_file_via_serial
    raise OSError(
OSError: Could not open file /main.py for writing, output:
Traceback (most recent call last):
  File "<stdin>", line 4, in <module>
OSError: [Errno 19] ENODEV

aivarannamaa, 2023-12-26:
Thank you for the report!

It looks like MicroPython could not open a file for writing. You could try manually doing `open("blaah.txt", "w")` in the REPL. Does it succeed?

Ativerc, 2025-01-14:
Just a heads up to anyone else reading this. I got the same error when trying to save a file on the ESP32 device. And once the following error happened, I would have to hard-reset and restart Thonny every single time.

```
PROBLEM IN THONNY'S BACK-END: Exception while handling 'write_file' (OSError: Could not open file /main.py for writing, output:
Traceback (most recent call last):
  File "<stdin>", line 4, in <module>
OSError: [Errno 19] ENODEV).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 1.
```

I was using thonny as well I suspected a bad board. I flashed the Micropython firmware multiple times using the following command to no avail: 

```
esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 460800 write_flash -z 0x1000 ESP32_GENERIC-20241129-v1.24.1.bin
```

`flash_id`:
```
esptool.py flash_id
esptool.py v4.8.1
Found 33 serial ports
Serial port /dev/ttyUSB0
Connecting.....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: bb:aa:xx:yy:xx:zz
Uploading stub...
Running stub...
Stub running...
Manufacturer: 68
Device: 4016
Detected flash size: 4MB
Flash voltage set by a strapping pin to 3.3V
Hard resetting via RTS pin...
```
One hint was Thonny wasn't able to show the filesystem. Tried getting the size of the filesystem by using the menu at the top right of the filesystem. it showed 0B in size and free space.

Tried the following:

```
import os
# Check if we can create the filesystem
try:
    os.VfsFat.mkfs(bdev)
except:
    print("Filesystem already exists")
```
I couldn;t save the script. So i had to run the script directly without saving.

Gave it some time.

Then did this:
```
import machine
machine.reset()
```

And it works. I dodn't save the log unfortunately. So, I guess this will work until I flash the system again.