thonny/thonny #3046: Same error as:  tkinter.TclError: couldn't connect to display ":10.0" #1816

technical-publishing, 2023-12-20:
Hello, I have to reopen this issue, because the same error occurs on my Raspberry Pi 3B. I have the Bullseye 32-bit OS installed. I have executed the tktest.py file (as recommended in issue #1816) and the same error occurs on the command line... could you please help me, I want to run a script with neopixel leds and therefore I have to run thonny in root mode. Thank you very much in advance.

aivarannamaa, 2023-12-26:
I now tried with Bookworm-based RPi OS and still can't reproduce this error. It might be worth trying a Raspberry Pi forum.

technical-publishing, 2023-12-26:
Dear Aivar,

 

thank you very much for your response. After several checks I am sure that is a problem of XRDP server and not of Thonny. BTW: Thank you very much for this great tool, all my students work with it.

 

Best regards and all the best for you for 2024!

Werner