thonny/thonny #3052: USB Serial : no access to RP PicoW board

molarex, 2023-12-28:
Working with Thonny 4.1.4, Windows 10 and Micropython on a RP PicoW everything goes well for a while; then a Backend-error occurs
and I can't get to the board - independently of the program because

   PROBLEM IN THONNY'S BACK-END: Internal error (serial.serialutil.SerialException: WriteFile failed (PermissionError(13, 'Das Gerät     erkennt den Befehl nicht.', None, 22))).
   See Thonny's backend.log for more info.
    You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.
    Process ended with exit code 0.

Even after resetting, deleting/renaming the COM-port, changing cables, even re-installing Thonny I get the same error immediately after connecting the board(s). The boards continue to work fine as WLAN-server for sensors, if independently powered, but I'm unable to change or modify their program.

I experienced this with two different boards

Here is the backend-log :

[backend.log](https://github.com/thonny/thonny/files/13787314/backend.log)

jimbo95-beep, 2024-01-04:
my pi pico does not show up in Mico Python mode it also will upload progammes code

aivarannamaa, 2024-02-24:
I don't know what causes this permission error and I don't think I could do anything about this in Thonny. I suggest asking advice from a MicroPython forum.

closed 2024-02-24 by aivarannamaa