thonny/thonny #3055: Font size in Variables pane v4.1.4

martybillingsley, 2024-01-04:
In older versions of Thonny (3.x), when the font size is changed (via Settings/Theme & Font), the font size in panes like Variables also changed. 

In v4.1.4, increasing the font size does not change the font size in the Variables pane (even after restarting Thonny).

aivarannamaa, 2024-01-07:
This change was made on purpose to make Thonny more like other IDEs.

Did you know that by changing "UI scaling factor" in "Tools -> Options -> General" you can change all sizes at once?

martybillingsley, 2024-01-08:
> This change was made on purpose to make Thonny more like other IDEs.
> 
> Did you know that by changing "UI scaling factor" in "Tools -> Options -> General" you can change all sizes at once?

I didn't know that; thanks for pointing it out! It'll help for the time being, although I really don't need the text outside the panes to be huge -- just need my students to be able to read the text in each of the panes. 

Unlike most IDEs, Thonny is a great teaching tool. Maybe the UI could reflect that???

aivarannamaa, 2024-01-08:
I'll add a configuration variable, which relates the font size of items in the variables panel (and file browser) to the editor font size.

martybillingsley, 2024-01-09:
> I'll add a configuration variable, which relates the font size of items in the variables panel (and file browser) to the editor font size.

Fantastic! Maybe some of the other panes too? Variables/Heap/Stack/Object Inspector are panes I use with my classes. I would think Program Tree could stay small; there's so much info there that large text would be unwieldy.

speedy-10, 2024-02-01:
I would appreciate variabel font size in "variables", too. This helps to teach python.

speedy-10, 2024-02-01:
> This change was made on purpose to make Thonny more like other IDEs.
> 
> Did you know that by changing "UI scaling factor" in "Tools -> Options -> General" you can change all sizes at once?
I'm using 'strg+ +" to increase font size in the editor. Perhaps it would be an idea to use 'strg+ + ' to increase the UI scale factor directly in "Tools -> Options -> General" ?

martybillingsley, 2024-02-01:
> I'm using actually 'strg+ +" to increase font size in the editor. Perhaps it would be an idea to use 'strg+ + ' to increase the UI scale factor directly in "Tools -> Options -> General" ?

Can you explain how to use strg++ to change the editor?
Thanks.

speedy-10, 2024-02-02:
As I wrote: open editor with code; press 'strg + +' and the font size is increasing (except  in "variables")