thonny/thonny #3058: WARNING: Secure coding is not enabled for restorable state!

caseminario, 2024-01-06:
I get this warning using turtle ..
2024-01-06 12:32:08.079 Python[13643:1383350] WARNING: Secure coding is not enabled for restorable state! Enable secure coding by implementing NSApplicationDelegate.applicationSupportsSecureRestorableState: and returning YES.

Here's my code causing the error ..
import turtle
turtle.forward(50)

macOS Sonoma Version 14.2.1
MacBook Pro 2.3 GHz 8-Core Intel Core i9
Thonny 4.1.4, Python 3.10.11, Tk 8.6.13

aivarannamaa, 2024-01-07:
You can ignore this warning. Next Thonny version comes with newer Python with this problem fixed.

closed 2024-01-07 by aivarannamaa

caseminario, 2024-01-08:
> You can ignore this warning. Next Thonny version comes with newer Python with this problem fixed.

Thanks Aivar! Any idea when the next Thonny version will be shipped? I have a Python course starting on Jan 16 and we start using Turtle heavily; if the new code isn't out yet, I will create a workaround.

aivarannamaa, 2024-01-08:
By 16th of January I may be able to release the first beta of the next version but you shouldn't count on this.

caseminario, 2024-01-08:
Ok, thanks for letting me know!

Jedikoops, 2024-06-06:
welp.

Zubair-codes1, 2024-09-18:
i cant ignore this issue its causing my code not to run