thonny/thonny #3063: Cannot select local micropython image file

brokenbrigde9000, 2024-01-09:
I wanted to flash a specific micropython image file on my ESP32-S3 (connected with a USB device, USB Single Serial @ /dev/ttyACM0), but the menu does not open. I can click the [Select local Micropython image...] option, but nothing happens.

Everything else in the application seems to work fine. Automatic flashing works as expected. I'm running linux, on the same system I did not have this problem under Windows 11.

Some system info:
Linux distribution: EndeavourOS
KDE Plasma Version 5.27.9
KDE Frameworks Version 5.112.0
Qtr Version: 5.15.11
Kernel version: 6.6-3arch1-1 (64-bit)
Graphics Platform X11

Any ideas how to have this menu pop up?

aivarannamaa, 2024-01-09:
Which Thonny version are you using?

brokenbrigde9000, 2024-01-10:
> Which Thonny version are you using?

sorry, forgot to add it: Thonny 4.1.3, Python 3.11.5, Tk 8.6.13