thonny/thonny #3064: Thonny not working

daz654, 2024-01-09:
Hello, I got home today and when attempting to open a python file in thonny i recieved the following error codes:
Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\tkinter\__init__.py", line 1921, in __call__
    return self.func(*args)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\base_file_browser.py", line 590, in on_double_click
    self.open_file(path)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\base_file_browser.py", line 1075, in open_file
    get_workbench().get_editor_notebook().show_file(path)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\editors.py", line 1050, in show_file
    editor = self.get_editor(filename, True)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\editors.py", line 1141, in get_editor
    return self._open_file(filename_or_id)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\editors.py", line 1123, in _open_file
    if editor._load_file(filename):
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\editors.py", line 208, in _load_file
    result = self._load_local_file(filename, keep_undo)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\editors.py", line 233, in _load_local_file
    source = fp.read()
OSError: [Errno 22] Invalid argument

Not too computer savvy, anybody have any ideas? thanks a lot

aivarannamaa, 2024-01-15:
Thank you for the report!

What's the name and path of the file you were trying to load?

aivarannamaa, 2024-09-11:
Please reopen if you still have this issue and can add more information.

closed 2024-09-11 by aivarannamaa